slic3r/Slic3r #4288: Import gcode file into Slic3r

theod, 2018-02-03:
### Version
Release Version 1.36.2-prusa3d

### Operating system type + version
OSX El Capitan

Hi,
this is a new feature request.

I would like to write hand writing curved letters on top of a model using my Prusa i3 mk2 printer as a pen.

one option could be to create a model of letters in my favorite 3D software, export it as svg file and then import it into Slic3r but in this case the slicing program will consider them as any volume and wont simply follow the hand writing path and the result will not be as fine as I imagine.

so my approach is to convert my hand writing svg letters into gcode using an Inkscape plugin as explain in the following tutorial (http://www.instructables.com/id/Make-Your-3D-Printer-Draw-Things/) but at this point I'm stuck as I can't find any way to import gcode file into Slic3r.

do I miss something ?

if not, it would be amazing to import such gcode trajectories as a layer and insert them on top of another already sliced model.

but maybe the solution is to merge gcode file myself and forget about the dream of doing it into a wysiwyg software ?

many thanks for your great work !

theod, 2018-02-03:
I need to say I don't want to write with a pen attached to my extruder as explain in the given tutorial ! I want to write hand writing letters with the material coming out from the extruder it self.

theod, 2018-02-03:
maybe a better approach would be to request to be able to import svg file and then slic3r would produce gcode depending of print, filament and printer settings ...

lordofhyphens, 2018-02-03:
This is probably done better outside of Slic3r, either processing the SVG directly and treating each line as a rectangle (generating an OpenSCAD file would likely be quickest) or modifying the draw plugin you already have.

There is an Inkscape plugin to generate OpenSCAD file from shapes already. OpenSCAD can then be used to make an STL (which is where you finally arrive at Slic3r). That may not be what you want, though.

If you have gcode, you should only need the E axis settings. Work out the volume of plastic for a unit length and speed of extrusion. The speed is pretty easy (mm/min is the unit in Gcode), flow math you can steal from the Slic3r manual. If you wanted to just experiment then take a calibration cube, turn off infill and only do a single perimeter at the width you want. That will give you a unit distance to simplify calculations.

 Modify the Inkscape plugin to produce this gcode. M Stick your desired start/end gcode on it.

closed 2018-02-03 by lordofhyphens

theod, 2018-02-03:
ok I'll work around the E axis setting solution. 
thanks for your quick reply !