slic3r/Slic3r #4289: Infill perimenters overlap mm input doesn't work

rockandsalt, 2018-02-04:
### Version
1.3.0-dev
1.2.9-1275-gb2d5cd97

### Operating system type + version
Ubuntu 64-bit

### Behavior
* It seems the mm option in infill/perimeters overlap doesn't work. When I write a simple number, there's no effect on the slic3r. 
* _Step to reproduce_
  * add simple cube stl
  * go "print-setting" -> "advanced" -> "infill/perimeters overlap"
  * write number with mm
* _Expected Results_
  * you should see the infill cross the perimeters. When I write the value in percent, the slic3r does the overlap. I want the input to be mm. 
  * [image]

* _Actual Results_
  * [image]

I am also willing to go and fix this. If this is a bug. I just need some pointers on where the infill parameters is processed. 

Cheers

rockandsalt, 2018-02-05:
So I dig some digging, turns out the feature works fine. But I need to write values in 10^6 range to see any effect when I use mm. Is there a reason why?

lordofhyphens, 2018-02-06:
10^6?

Ouch. I bet that it didn't scale it properly in the code where it is used.
There is a normal macro/function to scale/unscale in libslic3r.

Do a full text search for the parameter name (in the config) in the xs
directory to spot where it is used. Do another search to find the scale
function. Wrap the usage of the parameter with scale, rebuild, and see how
it goes ;)

Good find.

rockandsalt, 2018-02-09:
@lordofhyphens  Can you run slic3r with breakpoints? I'd like to put breakpoints in the c++ portion of it.

lordofhyphens, 2018-02-09:
You should be able to debug libslic3r with breakpoints; compile with -g and debug the perl exec.

Alternatively stick debug prints there.

rockandsalt, 2018-02-09:
@lordofhyphens  what is the coordinate system and the unit sli3r uses when it generates the infill. I put some print statement in FillRectilinear::_fill_single_direction. The points coordinate vary between -10^7 and 10^7 .

supermerill, 2018-08-03:
found why:
in layerregionconfig, there are 
```
f->endpoints_overlap = this->region()->config.get_abs_value("infill_overlap",
            (perimeter_spacing + scale_(f->spacing))/2);
```

it should be
```
f->endpoints_overlap = scale_(this->region()->config.get_abs_value("infill_overlap",
            (unscale(perimeter_spacing) + (f->spacing))/2));
```

masterjuggler, 2019-04-28:
Seems this is still an issue, where you need absurdly large absolute values to see any meaningful overlap, where relative values work as expected. Just seeing if there has been any change since this was opened.

lordofhyphens, 2019-04-28:
@masterjuggler you can give the change proposed in #4796 a try.

masterjuggler, 2019-04-28:
Building with that patch seems to resolve the issue, didn't see any strange behavior in my short test. 

Dunno if there's any procedure you'd like me to follow officially, but it looks good to me. Thanks for setting that up so quickly after my bump.

closed 2019-04-28 by lordofhyphens

lordofhyphens, 2019-04-28:
@rockandsalt does this sort the issue out for you?

rockandsalt, 2019-04-28:
yeah, this works

PR #4796 (2019-04-28): Apply fix proposed by @supermerill for #4289
Fixes #4289