slic3r/Slic3r #4290: [Feature Request] Multi-color support for SVG slicing (-> please help me do it myself)

N4rki, 2018-02-04:
### Version
1.2.9-1272-gdc9d7ed7

### Operating system type + version
Windows 10 Pro

### Description
* For my bachelor thesis which involves multi-material slicing for Inkjet 3D Printing it would be great to have different objects within one .3mf file to be sliced to a multicolor SVG. Each color in the SVG should represent one of the objects. Right now, Slic3r produces SVG layers that show all objects in white. I will add some pictures of what I mean tomorrow.
* I am more than happy to do the work myself. I also want to have a go at fixing bug [4283](https://github.com/alexrj/Slic3r/issues/4283). But even after spending days reading and trying different things, as well as consulting a friend who is majoring in Computer Science, I'm still having issues getting started developing on slic3r. 

### What works so far:
 * Compiling and running Slic3r from source
 * Compiling the [CLI only version with cmake](https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-Windows#compiling-slic3r-cli-only-limited---no-perl) (after switching the order of the parameters and following [this](https://stackoverflow.com/questions/3808775/cmake-doesnt-find-boost) in order for cmake to find the boost libraries. -D Boost_INCLUDE_PATH=C:\dev\boost_1_63_0 didn't work for me)

### My issues
 _These are somewhat separate but I didn't want to flood this forum with too many posts. Since they are all just ways to get to the same goal , finding a solution for one of them would effectively make the others void._
1. First off, when I change the text of the SVG header in SLAPrint.cpp, compile from source and do `slic3r.pl torus.stl --export-svg` it doesn't actually change the header of the  produced svg file to what I specified. In the source file I added "Test1 SLAPrint - n4rki" in the first line of the header but it somehow doesn't show up in the svg file. The header text is also found in SVG.cpp (twice), so just to be sure I changed it there too. But no difference. Here's my code: [SLAPrint + SVG code.zip](https://github.com/alexrj/Slic3r/files/1693384/SLAPrint.SVG.code.zip)
Why are these changes not showing up in the svg files after compiling?
2. Even if the above worked, it would still be a rather inefficient way to program. Since I don't need to work on the GUI for now, I thought working with the CLI only Version might be easier. I managed to compile it with cmake, but it doesn't seem to work:
   * `slic3r.exe donut.stl `returns "error: command not supported"
   * `slic3r.exe donut.stl --export-svg` crashes after a while and returns: 
```
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
```
The instructions on [here](https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-Windows#compiling-slic3r-cli-only-limited---no-perl) describe that it works in a limited way, but there are no details on what should and shouldn't work.

3. [Building with Visual Studio 2013](https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-Windows#building-with-visual-studio-2013): The 2013 Version is hard to come by and the download for the perl module patches is offline. Is this way still recommended? Anybody here who uses it or has experience with later VS Versions?
4. How do YOU develop for slic3r? Is there a good way to do it on Windows and be able to debug etc. or should I just stop messing around, use Linux and get aquainted with Emas + GDB?

As you can see I am a rookie at this - sorry for asking such basic questions! I'd really like to get started with this and I'm facing major roadblocks. Any help is much appreciated!

lordofhyphens, 2018-02-05:
I build on Windows with MinGW (using a customized Strawberry Perl available on the wiki) . You'll have to be savvy with GDB though.

My primary build system  (when I am not debugging windows specific issues) is Debian Linux. I suggest using perlbrew on a development system, but Slic3r works fine with system Perl. Just install local-lib from the packages.

My preferred editor on both is (g)Vim :)

lordofhyphens, 2018-02-05:
Add -g to your CXXFLAGS environment variable (or edit build.PL or the cmake directly).

N4rki, 2018-02-07:
@lordofhyphens Thanks, that helped a lot! And adding -g worked wonders for gdb :)

I'm happy to debug with gdb, but having some Vim integration for gdb with in-text highlighting would be a huge help. Do you use such a plugin when debugging on windows? I'm asking because I can't get [ConqueGdb](http://www.vim.org/scripts/script.php?script_id=4582) to work properly. The Conque Shell and gdb runs normally but without going through and highlighting the actual code in Vim.
But I guess I'll just go with gdbgui then.

Concering issue 1, I explored this further. So basically I type something into the SVG header text in SLAPrint.cpp (SLAPrint::write_svg) just to see if my changes are implemented into the programm through compiling. Should be a no-brainer, since it's kind of the definition of compiling from source, right? But yet when I compile with `perl build.pl` and then process an example with `perl slic3r.pl torus.stl --export-svg -layer-height 0.2` the header in the generated svg file is in its original state, not what I changed it to.
Weirdly, when I then build the CLI only version from the exact same source code with cmake and MinGW in a separate folder, then run `slic3r.exe torus.stl --export-svg -layer-height 0.2`, the generated svg file does have these changes in its header. 
Do you have any ideas what's going on here?

On that same token: How can I debug the whole application including perl? I got it into gdb with `gdb perl slic3r.pl` and then run with `set args slic3r.pl torus.stl --export-svg --layer-height 0.2` but it doesn't find any symbols in slic3r.pl. Do I have to change build.pl to debug mode as well (besides cmakelists) and if so, how?

lordofhyphens, 2018-02-08:
I try not to debug on Windows :)

Usually I don't need to go into detail on the perl side so my debug methods have been loading core dumps after the fact (as I am usually looking at crashes), or I am sticking diagnostic print statements into the source itself.

So I will try to have a better answer for you later.

lordofhyphens, 2018-02-08:
Conque indicates that it needs Python support in gVim to work. That also means that the correct bitness (32 vs 64bit). `:version` will tell you what it is and what options it is linked with.

I don't know if you've checked that support yet with `:echo has("python")` and related things yet, but it's a good idea to do so.
https://stackoverflow.com/questions/4718122/how-to-enable-python-support-in-gvim-on-windows

N4rki, 2018-02-09:
> Conque indicates that it needs Python support in gVim to work. That also means that the correct bitness (32 vs 64bit). :version will tell you what it is and what options it is linked with.
>
> I don't [...]

Yes I tried those things. My Vim had python. I went pretty deep down the rabbit hole with some of it. I though the fact that MinGW needs to have GDB might be the cause. Because I do have MinGW (from Slic3r Perl) and I do have GDB. I just wasn't so sure if my MinGW had gdb (and if that might matter).
The problem is that when getting MinGW through Slic3r Perl `mingw-get.exe install gdb` gave me a lock error although I ran the CLI as admin. 
So I deleted Slic3r Perl to start with a fresh plate when it comes to MinGW. Installed that the normal way, added gdb to it but got the exact same results. 

I'm trying to get gdbgui to work now which is a mess on windows too. I'm very close to convincing my boss to use a linux virtual machine which he was open to as a last resort but it would still be frustrating after having come so close to get a workflow going on windows.
Anyway thanks for the help, I'm sure I'll find a way.

If in the meantime, if anybody has any pointers for me on how to implement my feature that'd be cool, too! I not quite yet sure where exactly where I would take the color values of different objects in a .3mf from. I will update this with my findings.

lordofhyphens, 2018-02-09:
For looking at 3mf, look over the existing importer first. You may need to extend the existing code for that.

Ok, last sanity check: your gvim and python are both either 64 bit or 32bit right ?

The default mswindow build of gvim is 32bit.

N4rki, 2018-02-09:
Yes, I matched the bitness of gvim and python.

I did find the solution to get gdbgui working though: I had not only install normal MinGW but also add all packages in the installation process. Don't know which one made the difference but apparently not everything thats needed is included by default. Didn't help the ConqueGdb Vim plugin but I'm happy with switching to gdbgui.

Finally I can start working :)

lordofhyphens, 2018-02-09:
So the one wrinkle you may come across is anything that throws an exception; you need to make sure that SEH exceptions are being used. SJLJ causes hangs when exceptions percolate up to the Perl side.

I think the default exception model of gcc7 for mingw is SEH, but you should check your download to be sure.

It is the reason why I had rolled Slic3r perl (it is strawberry perl 5.24 with the gcc swapped out).

N4rki, 2018-02-13:
Thanks for mentioning that. This was indeed a problem, as I still need to be able to compile slic3r. I then tried it with [this](https://sourceforge.net/projects/mingw-w64/files/Toolchains%20targetting%20Win32/Personal%20Builds/mingw-builds/7.2.0/) MinGW install (tried x86_64-win32-seh and x86_64-posix-seh. But had no luck compiling the CLI only version of slic3r. In case you're interested:

<details> 
<summary> Click here for the CLI output</summary>

```
C:\Users\VB\Documents\Bachelorarbeit\Slic3r_new1\Slic3r\build>mingw32-make.exe
Scanning dependencies of target clipper
[  1%] Building CXX object CMakeFiles/clipper.dir/C_/Users/VB/Documents/Bachelorarbeit/Slic3r_new1/Slic3r/xs/src/clipper.cpp.obj
[  2%] Linking CXX static library libclipper.a
[  2%] Built target clipper
Scanning dependencies of target libslic3r
[  3%] Building CXX object CMakeFiles/libslic3r.dir/C_/Users/VB/Documents/Bachelorarbeit/Slic3r_new1/Slic3r/xs/src/libslic3r/BoundingBox.cpp.obj
[  4%] Building CXX object CMakeFiles/libslic3r.dir/C_/Users/VB/Documents/Bachelorarbeit/Slic3r_new1/Slic3r/xs/src/libslic3r/BridgeDetector.cpp.obj
[  6%] Building CXX object CMakeFiles/libslic3r.dir/C_/Users/VB/Documents/Bachelorarbeit/Slic3r_new1/Slic3r/xs/src/libslic3r/ClipperUtils.cpp.obj
[  7%] Building CXX object CMakeFiles/libslic3r.dir/C_/Users/VB/Documents/Bachelorarbeit/Slic3r_new1/Slic3r/xs/src/libslic3r/Config.cpp.obj
C:\Users\VB\Documents\Bachelorarbeit\Slic3r_new1\Slic3r\xs\src\libslic3r\Config.cpp: In member function 'void Slic3r::ConfigBase::save(const string&) const':
C:\Users\VB\Documents\Bachelorarbeit\Slic3r_new1\Slic3r\xs\src\libslic3r\Config.cpp:419:56: warning: unknown conversion type character 'F' in format [-Wformat=]
         strftime(buf, sizeof buf, "%F %T", gmtime(&now));
                                                        ^
C:\Users\VB\Documents\Bachelorarbeit\Slic3r_new1\Slic3r\xs\src\libslic3r\Config.cpp:419:56: warning: unknown conversion type character 'T' in format [-Wformat=]
[  8%] Building CXX object CMakeFiles/libslic3r.dir/C_/Users/VB/Documents/Bachelorarbeit/Slic3r_new1/Slic3r/xs/src/libslic3r/ExPolygon.cpp.obj
[  9%] Building CXX object CMakeFiles/libslic3r.dir/C_/Users/VB/Documents/Bachelorarbeit/Slic3r_new1/Slic3r/xs/src/libslic3r/ExPolygonCollection.cpp.obj
[ 10%] Building CXX object CMakeFiles/libslic3r.dir/C_/Users/VB/Documents/Bachelorarbeit/Slic3r_new1/Slic3r/xs/src/libslic3r/Extruder.cpp.obj
[ 12%] Building CXX object CMakeFiles/libslic3r.dir/C_/Users/VB/Documents/Bachelorarbeit/Slic3r_new1/Slic3r/xs/src/libslic3r/ExtrusionEntity.cpp.obj
[ 13%] Building CXX object CMakeFiles/libslic3r.dir/C_/Users/VB/Documents/Bachelorarbeit/Slic3r_new1/Slic3r/xs/src/libslic3r/ExtrusionEntityCollection.cpp.obj
[ 14%] Building CXX object CMakeFiles/libslic3r.dir/C_/Users/VB/Documents/Bachelorarbeit/Slic3r_new1/Slic3r/xs/src/libslic3r/Fill/Fill.cpp.obj
[ 15%] Building CXX object CMakeFiles/libslic3r.dir/C_/Users/VB/Documents/Bachelorarbeit/Slic3r_new1/Slic3r/xs/src/libslic3r/Fill/Fill3DHoneycomb.cpp.obj
[ 17%] Building CXX object CMakeFiles/libslic3r.dir/C_/Users/VB/Documents/Bachelorarbeit/Slic3r_new1/Slic3r/xs/src/libslic3r/Fill/FillConcentric.cpp.obj
[ 18%] Building CXX object CMakeFiles/libslic3r.dir/C_/Users/VB/Documents/Bachelorarbeit/Slic3r_new1/Slic3r/xs/src/libslic3r/Fill/FillHoneycomb.cpp.obj
[ 19%] Building CXX object CMakeFiles/libslic3r.dir/C_/Users/VB/Documents/Bachelorarbeit/Slic3r_new1/Slic3r/xs/src/libslic3r/Fill/FillPlanePath.cpp.obj
C:\Users\VB\Documents\Bachelorarbeit\Slic3r_new1\Slic3r\xs\src\libslic3r\Fill\FillPlanePath.cpp: In function 'Slic3r::Point Slic3r::hilbert_n_to_xy(size_t)':
C:\Users\VB\Documents\Bachelorarbeit\Slic3r_new1\Slic3r\xs\src\libslic3r\Fill\FillPlanePath.cpp:142:9: warning: variable 'dirstate' set but not used [-Wunused-but-set-variable]
     int dirstate = (ndigits & 1) ? 0 : 4;
         ^~~~~~~~
[ 20%] Building CXX object CMakeFiles/libslic3r.dir/C_/Users/VB/Documents/Bachelorarbeit/Slic3r_new1/Slic3r/xs/src/libslic3r/Fill/FillRectilinear.cpp.obj
[ 21%] Building CXX object CMakeFiles/libslic3r.dir/C_/Users/VB/Documents/Bachelorarbeit/Slic3r_new1/Slic3r/xs/src/libslic3r/Flow.cpp.obj
[ 23%] Building CXX object CMakeFiles/libslic3r.dir/C_/Users/VB/Documents/Bachelorarbeit/Slic3r_new1/Slic3r/xs/src/libslic3r/GCode.cpp.obj
[ 24%] Building CXX object CMakeFiles/libslic3r.dir/C_/Users/VB/Documents/Bachelorarbeit/Slic3r_new1/Slic3r/xs/src/libslic3r/GCode/CoolingBuffer.cpp.obj
C:\Users\VB\Documents\Bachelorarbeit\Slic3r_new1\Slic3r\xs\src\libslic3r\GCode\CoolingBuffer.cpp: In member function 'std::__cxx11::string Slic3r::CoolingBuffer::flush()':
C:\Users\VB\Documents\Bachelorarbeit\Slic3r_new1\Slic3r\xs\src\libslic3r\GCode\CoolingBuffer.cpp:126:25: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     if (this->_layer_id < gg.config.disable_fan_first_layers)
         ~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\VB\Documents\Bachelorarbeit\Slic3r_new1\Slic3r\xs\src\libslic3r\GCode\CoolingBuffer.cpp:132:82: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     if (!gg.config.cooling || gg.config.bridge_fan_speed == 0 || this->_layer_id < gg.config.disable_fan_first_layers) {
                                                                  ~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[... cut ...]
I did find a solution though: It seems that I can use the custom MinGW from Slic3r Perl and the normal installation in parallel. I just have to 'specify' which to use when. Since the normal MinGW is in C:/mingw64 (which is in PATH) I just change that folder to mingw64_1 whenever I want to compile Slic3r and change C:/Strawberry (which has the custom MinGW and is in PATH too) to Strawberry_1 whenever I want to debug with gdbgui. So whatever is not needed will not be available via PATH. This really isn't very elegant but doesn't take much time and I'll take it for now.

N4rki, 2018-03-27:
The initial goal I specified in this thread has been accomplished. At first I was using the 3mf specific part_number that is already read from 3mf source files and parsed that as a materialID. This ID is then used to determining the color of each polygon in SVG Slicing (via Window -> DLP Projector -> Export SVG). 
Even more so, my fork is now utilizing the "Extruder" setting of each volume for that so the user can set it from the Object Settings window. 

_By the way, when my project is finished I am more than happy to discuss which parts might be of value to Slic3r and what I'd have to possibly change in order to make it more useful and fitting with the general programming guidelines that should be followed. I would love to contribute something!_

Until then, there is another change I'd like to implement: It would be great if the produced SVG is not sized according to the bounding box of all ModelObjects, but instead takes the bed shape as size.

So this would be the `const Sizef3 size` parameter in `SLAPrint::write_svg`.

The bed_shape ConfigOption is contained in `PrintConfigDef` (as well as `class PrintConfig : public GCodeConfig`).

What I am not quite sure of is whether SLAPrintConfig already holds the bed_shape ConfigOption (because it extends SaticPrintConfig). Checking via `this->config.has("bed_shape")` in `SLAPrint::write_svg` returns false. Is there another way to directly grab this parameter from the PrintConfig?

**Alternatively**, a bed_shape configoption (or bed_size for that matter) would have to be added to the SLAPrintConfig and specified in the constructor in SLAPrintOptions.pm, right? So after having added `ConfigOptionPoints bed_shape;` to SLAPrintConfig, how exactly do I grab bed_shape from PrintConfig in SLAPrintOptions.pm?

Here is the start of SLAPrintOptions.pm. Would this be the right way and if so, can somebody complete the last call for me?
```
sub new {
    my ($class, $parent) = @_;
    my $self = $class->SUPER::new($parent, -1, "SLA/DLP Print", wxDefaultPosition, wxDefaultSize);
    
    $self->config(Slic3r::Config::SLAPrint->new);
    $self->config->apply_dynamic(wxTheApp->{mainframe}->{plater}->config);
    
    # Set some defaults
    $self->config->set('infill_extrusion_width', 0.5)  if $self->config->infill_extrusion_width == 0;
    $self->config->set('support_material_extrusion_width', 1)  if $self->config->support_material_extrusion_width == 0;
    $self->config->set('perimeter_extrusion_width', 1) if $self->config->perimeter_extrusion_width == 0;
    $self->config->set('bed_shape', <???>);
```
I thought I'd ask this here because it might be very easy for somebody who is more familiar with the config structure and perl.

N4rki, 2018-05-08:
I somehow got it to work :) I would say this can be closed now...

closed 2018-07-09 by lordofhyphens