slic3r/Slic3r #4291: Intermitant, Significant Slowdown of all Geometries other than Support

DocTrucker, 2018-02-08:
### Version
Slic3r 1.3 dev 1590

### Operating system type + version
Windows 7 64 bit

### Behavior
I'm trying to print a z-axis anti backlash nut holder in PET-G. Most layers are around 1min, but others suddenly jump to 10min. I can't see any obvious trigger for it on the geometry and it doesn't appear to slow down the supports.

Part was left centre of the build volume, but rotated 90 degrees. I'm seeing some feedrates that are in the region of 400, which is below what the auto cooling should slow them down to. I've not tried repeat builds with different setting but still seeing the 400ish feed rate with auto cooling disabled and supports not used.

[AntiBacklashZNutHolder.zip](https://github.com/alexrj/Slic3r/files/1707101/AntiBacklashZNutHolder.zip)

lordofhyphens, 2018-02-08:
Is autospeed on?

lordofhyphens, 2018-02-08:
Loaded the file and yes, autospeed is on. Autospeed tries to keep the flow rate even (same pressure) so when you have a small/thin extrusion everything else tends to get done relative to that in the same layer.

Something smarter is probably called for here, like gradually increasing speed/pressure for a series of short moves so you can have more than one flow rating for a layer. Unfortunately getting that number right depends on a lot of physical factors that change from plastic to plastic (it'd be based primarily on the viscosity, which is different at different temperatures and between different blends).

I suspect the allowable margin of error isn't too bad though. 

The current workaround for your situation though is to disable autospeed.

DocTrucker, 2018-02-10:
Thanks for the feedback. It looks like I had got the wrong end of the stick with regards to what Autospeed was about and it certainly needs some better controls. I set the speeds to similar values to what were being used on the whol and the build time dropped from ~5hrs to ~3hrs.

Background: I've been using TPU recently and found it a very slow material to process though a 0.4mm nozzle, E3D V6 with Titan Extruder driven by their slim line motor. I'd found when I pushed beyond the point of linear extrusion I could push up the extrusion multiplier until the single wall thicknesses were correct but running at the same speed for larger wall thicknesses was a no-hoper as the flow rate was proportionally higher when already running on the ragged edge. I used autospeed as I thought it would correct feedrates to maintain a maximum volumetric rate. What your saying makes sense, but the end result in this case is very visible changes in surface quality, and not exactly what I understood.

ForsakenNGS, 2018-03-08:
Just a dumb suggestion, how about adding a minimum speed setting for autospeed (in mm/s not volumetric) which prevents the whole layer to drop below that given speed? The small extrusions would of course still be printed below the minimum speed.

I'm aware this is not the desired final soution, but as of now autospeed is just not usable for most parts and this option should be very easy to implement. Like that it would at least be viable to use autospeed until a better solution for this problem is found and implemented, which I'm pretty sure will be quite a while.