slic3r/Slic3r #4293: Changing print settings to other one is ignoring the layer height when slicing

mag00-br, 2018-02-09:
### Version
Slic3r-master.2018.02.08.1593.05461b8.64bit

### Operating system type + version
Windows 10 Home x64

### Behavior
With any model in the screen, I click in preview and it slices the part correctly. Then I go to print setting pull down menu and change to any other configuration.
After that I click in preview again and it slices the model with all new configuration of new print setting but the new layer height. It always use the previous layer height. 
And exporting this slice still maintain the wrong layer height.

As you can see in below image the layer height is with 0.4 mm and the slicing bar shows 0.25 mm (in bottom of image).
[image]

This issue is occurring in DEV since november 2017 (or even before it).

#### STL/Config (.ZIP) where problem occurs
[polvinho.zip](https://github.com/alexrj/Slic3r/files/1709369/polvinho.zip)

platsch, 2018-02-13:
I'm unable to reproduce the problem.
What values do you use for Min / Max layer height in the Printer Settings tab? I noticed that this (correctly) limits the layer height but doesn't raise any warning if you set an out of range value
If that's not the reason: could you provide a set of exported configurations for further investigation?

mag00-br, 2018-02-13:
I've never changed these values.
It's showing 0.15 and 0.30.
I believe that this Min/Max layer configuration is not the problem, because after preview, if I noticed that the layer height is not with the correct value, then I change the value to other one and return it to value that I want and it "correct" the problem.
I've put the exported configurations inside file "polvinho.zip" of original post. Or you need something else?

platsch, 2018-02-23:
Still unable to reproduce.
What's your OS? Which exact version of Slic3r is affected? I realize that you mentioned the issue is present at least since November 2017 but I'd still like to test the exact same version.
If it is related to a certain OS version, maybe someone else can reproduce the problem.

You are correct, the Min / Max layer height from the extruder settings does not limit the layer heights except for adaptive layer generation or if `z full steps/mm` is used.
@lordofhyphens any opinion on this side-problem? I think it is currently inconsistent, but if we apply the extruder limits for static slicing I expect many people being confused. There should be at least a warning.

You are aware that background processing was disabled in your config? I had to trigger slicing manually after changing the preset.

mag00-br, 2018-02-23:
My OS is the Windows 10 Home x64.

I'm using development slic3r versions since June of last year (2017).

I've deactivated the background processing since forever.

I've started to notice this issue since November 2017 (or before), so, basically almost every dev version that was available since November 2017 have this issue, and I noticed it from 2 distinct computers, with 2 distinct OS.

The other computer was using Windows 8 Pro x64.

-----------------------------------------------------------------------------------------------------------

Other things that I can add to (maybe) help to reproduce the problem:
- This is my preferences
[image]

- I use most common parameters on right panels, due to I used to change them constantly. So I've marked the checkbox inside the **Print settings** to them always appear in the screen
[image]

- I've some predefined configurations saved to help me change uncommon settings a bit more quickly.
[image]
[image]
[image]

- All my print settings are forcing all types of extrusion widths be the same width. Doing this, I forcing slic3r to ignore the nozzle width in the **Printer settings**. (This makes me having some problems with bridges and supports, but it's not the point here.)
[image]

-----------------------------------------------------------------------------------------------------------

To end all this, I'll print a step-by-step of problem.
- Clear work area
[image]

- Model imported
[image]

- Preview with current settings already loaded (layer height with 0.3 mm)
[image]

- First layer (0.3 mm as configured)
[image]

- Second layer (0.6 mm, adding the first to the math) and all others are respecting the 0.3 mm each
[image]

- Now I've changed the Print settings to other pre configured one, that uses 0.4 mm as layer height
[image]

- Clicking in preview it shows the layers
[image]

- Now, the first layer that should be 0.4 mm is with 0.3 mm yet
[image]

- The second that should be 0.8 mm is with 0.6 mm, and every other layer is with 0.3 mm instead of 0.4 mm as this Print setting is showing in the screen
[image]

---------------------------------------------------------------------------------------------------

So, to make sure that the GCODE be generated correctly I need to change the layer height to another size, and then back to the size I really want to again.

Which brings us to the other problem that I posted too, that making the preview stop working correctly, but this issue have it own topic.

I believe that that's pretty much all the history, but if you need some other information, please ask.

Thanks.

mag00-br, 2018-04-05:
This issue was solved with correction made in issue #4364.

closed 2018-04-05 by mag00-br