slic3r/Slic3r #4294: Feature request: manually set small radius (perimeter) for other parameter

ArtemKuchin, 2018-02-09:
Multiple parameter use "small radius" to take effect. For example, there is separate setting for small perimeter speed. But small perimeter radius is hard coded. Please, add a manual setting for small radius perimeters. The idea is that small radius is different for different mechanics. For better quality or lower end machines small radius is probably good to set a lot higher.

gennartan, 2018-11-09:
Hello, I got this!!
I just need to know where I should add this option in the slicer. Do you think under "vertical shell" in the page "Layers and perimeters" or under the option "small perimeter speed" in the page "speed"
The small perimeter length option is only used for the computation of the speed for the small perimeter, that is why I would place this option in the speed section.

ArtemKuchin, 2018-11-09:
I think logical place for it is the same tab where the speed for small radius is set, so everything willl be in one place.  So, i think it should be on the "speed" tab.

gennartan, 2018-11-10:
Thank you for this confirmation.
I created a new pull request at https://github.com/slic3r/Slic3r/pull/4581