slic3r/Slic3r #4295: How to add new entry to GCodeConfig class and be able to call it in PresetEditor.pm?

rockandsalt, 2018-02-11:
I am trying to add a couple of new entry to the Printer Setting page. 
[link to my commit](https://github.com/Multi-Scale-Additive-Manufacturing-Lab/Slic3r/commit/e3972a34a1df796e40cf135a0ff7fcda8861e6f6)
Honestly at first it seems straight forward. I add the a new config entry to the GCodeConfig class or to a standalone class. Then I call  OptionsGroup::append_single_option_line in PresentEditor.pm to add the entry to the page. 

ex:
PrintConfig.hpp
```
class GCodeConfig : public virtual StaticPrintConfig
{
      \\new entry
      ConfigOptionFloat               start_height;
      
      virtual ConfigOption* optptr(const t_config_option_key &opt_key, bool create = false) {
             OPT_PTR(start_height);
      }
}
```
PrintConfig.cpp
```
    def = this->add("start_height", coFloat);
    def->label = "Start Height";
    def->sidetext = "mm";
    def->default_value = new ConfigOptionFloat(0);
```

But when I call OptionsGroup::append_single_option_line in PresentEditor.pm : 
`$optgroup->append_single_option_line('start_height');`

it throws the error : `No start_height in ConfigOptionsGroup config`.

I don't seem to be doing anything different than for example the config `gcode_flavor` in the  GCodeConfig class. 

Am I missing anything?

platsch, 2018-02-12:
Did you add it to the respective `sub options` in PresetEditor.pm?
There is one for each tab, e.g:

```
sub options {
    return qw(
        ...
        start_height
    )
```

rockandsalt, 2018-02-12:
@platsch, I actually didn't. Thanks for pointing it out. Do you know who is calling this? I am not seeing where it is called.

lordofhyphens, 2018-02-12:
https://github.com/alexrj/Slic3r/wiki/Code%3A-Adding-new-option-to-GUI-CLI

It is used later during the menu setup code.

rockandsalt, 2018-02-12:
Thanks that was helpful

closed 2018-02-12 by rockandsalt