slic3r/Slic3r #4296: Background Color of Layers (Tab)

foreachthing, 2018-02-12:
### Version
1.3 beta

### Issue
I'm working on "solarizing" (http://ethanschoonover.com/solarized) Slic3r and can't find where I need to adjust the background color of the Layers tab. Any hint is appreciated.
Everything else was pretty "easy" - if you know/knew Perl! Still work in progress: https://github.com/foreachthing/Slic3r/tree/Solarized

### Screenshots
Preview Tab:
[image]

Layers Tab:
The only thing I'm missing, is the background color. Everything else I got (not visible here).
[image]

lordofhyphens, 2018-02-13:
Take a look at 3DScene.pm. If you want to port the C++ 3DScene from the Prusa fork then I'd happily accept it.

lordofhyphens, 2018-02-13:
I'd also accept a PR for solarizing if you can make it an optional change in the preferences.

foreachthing, 2018-02-13:
Yes, in 3DScene, I was able to change the background for **3D** and **Preview**. Then in 2d.pm I could change the **2D** background. But the **Layers** ... !?
Ok... I need to take a look at the Prusa fork first. And yes, it's optional already ;-)

foreachthing, 2018-02-14:
I don't know .... I can't find it. I got the slider backcolor on **Layers** changed, though. 
It's like, it's not there for the grasp.
Does anybody see where I can change the background color of **Layers**?

lordofhyphens, 2018-02-24:
The Layers tab is drawn in 3D mode from my investigations. 

You are interested in the `glClearColor` call in `sub Render` in 2DToolpaths.pm.

For `glClearColor(0.3, 0.4, 0.1, 0);`
[image]

lordofhyphens, 2018-02-24:
http://prideout.net/archive/colors.php for your reference.

lordofhyphens, 2018-02-24:
You will also need to apply the same color to the individual line drawings.

closed 2018-02-26 by lordofhyphens