slic3r/Slic3r #4299: Feature Suggestion - Fixed inter-extrusion spacing

MyPrinter7, 2018-02-14:
Hi all!

Thanks for your amazing work!

I would like to make the suggestion of Fixed inter-extrusion spacing. This is something that would be invaluable to scientists around the globe working on tissue engineering projects.

We need to be able to select (and for each different object/modifier) for example 1.5mm track spacing so as to create repeatable structures for our research. This will assist in mechanical analysis, generating consistent porosity and in many other aspects. 

I would be happy to try to help or discuss further so please get in touch! 

Cheers!

MyPrinter7, 2018-02-20:
Hi guys, is this a possibility? Is the program setup so fundamentally differently that this wouldn't be possible?

lordofhyphens, 2018-02-20:
It is probably feasible. Nobody has needed it yet, though.

You could go after it as a special form of infill or through the Flow calculations. 

I would probably take the Infill route myself.

lordofhyphens, 2018-02-20:
Study the usage through SupportMaterial.pm (support fill uses Fill with some of the specs you are interested in).

MyPrinter7, 2018-02-20:
Ah yes I can see that support material spacing is an option - It would be absolutely incredible for our work to be able to translate this into regular infill spacing. I'll have a look through and let you know how I go - thanks for the pointer @lordofhyphens

MyPrinter7, 2018-02-20:
OK I think I have found the initial entry here under process layer...
my $support_spacing = $self->object_config->support_material_spacing + $flow->spacing;
    my $support_density = $support_spacing == 0 ? 1 : $flow->spacing / $support_spacing;

MyPrinter7, 2018-02-20:
Under GCODE I can see:

        {
            my $flow = $region->flow(FLOW_ROLE_INFILL, $layer_height, 0, 0, -1, $first_object);
            my $vol_speed = $flow->mm3_per_mm * $region->config->get_abs_value('infill_speed');
            $vol_speed = min($vol_speed, $self->config->max_volumetric_speed) if $self->config->max_volumetric_speed > 0;
            printf $fh "; infill extrusion width = %.2fmm (%.2fmm^3/s)\n",
                $flow->width, $vol_speed;}

But i'm guessing that the flow setup for infill would happen in Object.pm?
Is this right?