slic3r/Slic3r #4306: Latest automated builds are incomplete

foreachthing, 2018-02-19:
### Version
https://dl.slic3r.org/dev/win/
i.e.: Slic3r-master.2018.02.18.1602.5f28ee4.32bit.zip.filepart

### Operating system type
Windows

### Behavior
* Latest automated builds are incomplete
* if downloaded, extraction is followed by a disc error.

[image]

lordofhyphens, 2018-02-19:
Probably out of room. I will go clean things up in the branch dir.

lordofhyphens, 2018-02-21:
Did it sort itself out?

foreachthing, 2018-02-21:
I guess! There's a build from 2018-02-20...

foreachthing, 2018-02-23:
Nope! The issue still persists:
[image]

lordofhyphens, 2018-02-24:
Should be fine now.

lordofhyphens, 2018-02-24:
I've also kicked off a set of builds to make sure the last commit is available for download.

closed 2018-02-24 by lordofhyphens

lordofhyphens, 2018-02-25:
Yep, builds are good to go now.