slic3r/Slic3r #4307: Feature request: Place prime/wipe tower inside object

mrvn, 2018-02-19:
When printing with multiple extruders, especially diamond hotends, a color change needs to prime or wipe the extruder before the color flows cleanly. Slic3r supports a wipe tower that is placed at the side on the plate. But wouldn't it be even better to place that tower inside the object in place of infiill?

In many cases there is a lot of interior space in the object so a place can be found to wipe the nozzle for each layer. The amount of infill might be enough to change colors for a diamond hotend and if not then the infill can be increased or a region can be solid filled so enough material is extruded to prime the next color.

lordofhyphens, 2018-02-20:
Are you confusing Slic3rPE with Slic3r?

mrvn, 2018-02-21:
no

TheSin-, 2019-05-22:
if you aren't can you tell me how to enable this magically wipe tower cause me and LOTS of others would LOVE to have that in Slic3r (which BTW does not have this option)