slic3r/Slic3r #4310: [Bug] Slicing to 3MF to SVG scatters parts all over platform

N4rki, 2018-02-19:
### Version
1.2.9-1272-gdc9d7ed7

### Operating system type + version
Windows 10 Pro

### Description
When slicing a multi-part 3MF file to SVG the different parts are not positioned as they should in relation to another but scattered all over the platform. Additionally each slice of each part is in it's own layer.

### Reproduce
We take this example from microsoft's 3mf repository: [Train Engine.zip](https://github.com/alexrj/Slic3r/files/1737849/Train.Engine.zip)
which looks like this: 
[image]
Slicing it via File -> Slice to SVG... produces the following SVG: [Train Engine_svg.zip](https://github.com/alexrj/Slic3r/files/1737867/Train.Engine_svg.zip)
Here are the first 9  layers to illustrate what happens:
[image]
(read from left to right, top to bottom)

I don't really mind the fact that each object is in a separate layer, it even makes it a but easier for my application. But they should be positioned correctly on the platform

I noticed that this is handled differenty when slicing through Window -> DLP Projector -> Export SVG.
Likely because this is done in SLAPrint.cpp and not in print.pm like when going via File -> Slice to SVG.

Therefore I will try to go through the Window option for now and leave this Bugreport here in case other people have similar issues.

lordofhyphens, 2018-02-20:
SLAPrint.cpp has some issues with it; I have not had time to go digging into it.

lordofhyphens, 2018-02-25:
@Samir55 any ideas if there is anything weird going on with 3mf itself?

Patlax, 2023-12-11:
Not only when slicing a multi-part 3MF file to SVG, but also when slicing any other multi-part Object (3D-Object or stl), the different parts are not positioned as they should in relation to another but scattered all over the platform. 

Additionally, when slicing one or multiple objects, only one contour of all detected contours in layer->slices is written to the svgs, the infills are ignored fully. Why is that? Is this also a bug? Is there an option to add the infills to the 3d-printer-svg-file? I couldn´t find a suitable method for this yet.