slic3r/Slic3r #4311: [Bug] DLP Projector -> Export SVG produces incomplete SVG

N4rki, 2018-02-19:
### Version
1.2.9-1272-gdc9d7ed7

### Operating system type + version
Windows 10 Pro

### Description
When slicing any filetype to SVG via `Window -> DLP Projector -> Export SVG` the generated SVG is always missing the last two or three lines of xml text.

### Reproduce
Taking this example STL file: [Train Engine wheels.zip](https://github.com/alexrj/Slic3r/files/1738092/Train.Engine.wheels.zip)

Slicing it as stated above produces the following svg: 
[Train Engine wheels_svg.zip](https://github.com/alexrj/Slic3r/files/1738094/Train.Engine.wheels_svg.zip)
The incompleteness is only at the very bottom of the file. When we open it with an editor we see the following in the last section:
```
(...)
<g id="layer9" slic3r:z="4.1000" slic3r:slice-z="3.9000" slic3r:layer-height="0.4000">
		<path d="M 58.4608 17.4357 59.3813 17.2241 60.1623 16.9819 60.3994 16.8998 61.1425 16.5554 61.3603 16.4296 62.049 15.9897 62.2386 15.8251 62.863 15.2962 63.5103 14.6626 63.5619 14.5971 64.03 13.817 64.1452 13.5937 64.5143 12.8637 64.7829 12.0909 64.8536 11.85 65.0168 11.0492 65.0407 10.7991 65.0951 9.98179 65.0407 9.1661 65.0167 8.91643 64.8536 8.11556 64.7828 7.87472 64.5142 7.10188 64.399 6.87867 64.0299 6.14864 63.5616 5.36845 63.5102 5.30316 62.8628 4.6694 62.6734 4.50503 62.0488 3.97602 61.3599 3.536 61.1424 3.41035 60.3991 3.06592 60.1622 2.98393 59.3907 2.68822 59.3085 2.67646 58.378 2.51826 57.5076 2.54302 57.257 2.55499 56.4429 2.6374 56.199 2.69653 55.403 2.88608 54.6399 3.17944 54.4119 3.28362 53.6948 3.67752 53.4903 3.82289 52.8311 4.30742 52.1248 4.91483 52.0675 4.97498 51.562 5.70034 51.4261 5.91157 50.9584 6.59687 50.9274 6.6741 50.5583 7.54234 50.3804 8.39077 50.3328 8.63726 50.2241 9.44801 50.2241 9.69911 50.2241 10.5181 50.3329 11.3288 50.3804 11.5753 50.5979 12.3646 50.6913 12.5978 51.0093 13.3508 51.4262 14.0544 51.5622 14.2658 52.0697 14.9084 52.243 15.0902 52.8313 15.6586 53.036 15.8041 53.6951 16.2885 54.412 16.6823 54.6403 16.7866 55.4032 17.0798 55.6474 17.139 56.4432 17.3285 57.2571 17.4109 57.508 17.4228 58.3782 17.4476 z M 57.4628 13.5364 56.8975 13.4597 56.8086 13.4378 56.2665 13.2613 55.7636 12.992 55.6825 12.9496 55.2367 12.5947 55.1685 12.534 54.7939 12.1042 54.7419 12.0291 54.4495 11.5386 54.2479 11.0054 54.2154 10.9198 54.1138 10.3581 54.1027 10.2673 54.1027 9.69842 54.2044 9.13663 54.2154 9.0461 54.4169 8.51285 54.4494 8.42731 54.7419 7.93687 54.7939 7.86173 55.1684 7.43189 55.6142 7.07698 55.6824 7.01635 55.7634 6.97391 56.2664 6.70459 56.8085 6.52802 56.8973 6.50615 57.4627 6.42945 58.0316 6.45442 58.1228 6.45441 58.6792 6.58103 58.768 6.60289 59.2923 6.8276 59.3733 6.87003 59.8495 7.18393 59.9177 7.24456 60.3297 7.63869 60.665 8.10059 60.717 8.17574 60.9631 8.68937 60.9956 8.77492 61.1472 9.3242 61.1989 9.89132 61.21 9.98186 61.1582 10.5508 61.1472 10.6415 60.9956 11.1908 60.9631 11.2764 60.717 11.79 60.3818 12.2519 60.3298 12.3271 59.9178 12.7212 59.8496 12.7819 59.3734 13.0957 59.2924 13.1382 58.7681 13.3629 58.6793 13.3848 58.1229 13.5114 57.554 13.5364 z " style="fill: white; stroke: black; stroke-width: 0; fill-type: evenodd" slic3r:area="-0.0021" />
		<path d="M 34.5909 17.4361 35.5114 17.2245 36.2924 16.9823 36.5296 16.9002 37.2726 16.5558 37.4904 16.43 38.1791 15.9901 38.3687 15.8255 38.9931 15.2966 39.6404 14.663 39.6919 14.5975 40.1601 13.8174 40.2753 13.5941 40.6444 12.8641 40.913 12.0913 40.9837 11.8504 41.1469 11.0496 41.1708 10.7995 41.2252 9.98221 41.1708 9.16651 41.1468 8.91684 40.9837 8.11597 40.9129 7.87513 40.6443 7.10228 40.5291 6.87906 40.16 6.14905 39.6917 5.36886 39.6403 5.30357 38.9929 4.66981 38.8035 4.50543 38.1789 3.97643 37.49 3.53641 37.2725 3.41077 36.5292 3.06634 36.2924 2.98434 35.5208 2.68863 35.4386 2.67687 34.5081 2.51867 33.6377 2.54342 33.3871 2.55539 32.573 2.63781 32.3291 2.69694 31.5331 2.88649 30.77 3.17985 30.542 3.28403 29.8251 3.67793 29.6206 3.8233 28.9614 4.30783 28.2551 4.91523 28.1978 4.97538 27.6923 5.70075 27.5564 5.91198 27.0887 6.59729 27.0577 6.67451 26.6886 7.54275 26.5107 8.39118 26.4631 8.63767 26.3544 9.44841 26.3544 9.69951 26.3544 10.5185 26.4632 11.3292 26.5107 11.5758 26.7282 12.365 26.8216 12.5982 27.1396 13.3512 27.5565 14.0548 27.6925 14.2662 28.2 14.9088 28.3733 15.0906 28.9616 15.659 29.1663 15.8045 29.8254 16.2889 30.5421 16.6827 30.7704 16.787 31.5333 17.0803 31.7775 17.1395 32.5733 17.3289 33.3872 17.4113 33.6381 17.4233 34.5083 17.448 z M 33.0276 13.4601 32.9387 13.4382 32.3966 13.2617 31.8937 12.9924 31.8126 12.95 31.3668 12.5951 31.2986 12.5344 30.924 12.1046 30.8721 12.0295 30.5796 11.5391 30.378 11.0058 30.3455 10.9203 30.2439 10.3585 30.2328 10.2677 30.2328 9.69883 30.3345 9.13705 30.3455 9.04652 30.547 8.51326 30.5795 8.42772 30.872 7.93727 30.924 7.86213 31.2985 7.4323 31.7443 7.07738 31.8125 7.01675 31.8935 6.97431 32.3965 6.70501 32.9386 6.52843 33.0274 6.50656 33.5928 6.42986 34.1617 6.45482 34.2529 6.45482 34.8093 6.58144 34.8981 6.6033 35.4224 6.82801 35.5034 6.87044 35.9796 7.18434 36.0478 7.24497 36.4598 7.6391 36.7951 8.10101 36.8471 8.17615 37.0932 8.68977 37.1257 8.77531 37.2773 9.32461 37.329 9.89173 37.3401 9.98227 37.2883 10.5512 37.2773 10.6419 37.1257 11.1912 37.0932 11.2768 36.8471 11.7904 36.5119 12.2523 36.4599 12.3275 36.0479 12.7216 35.9797 12.7823 35.5035 13.0961 35.4225 13.1386 34.8982 13.3633 34.8094 13.3852 34.253 13.5118 33.6841 13.5368 33.5929 13.5368 z " style="fill: white; stroke: black; stroke-width: 0; fill-type: evenodd" slic3r:area="-0.0021" />
		<path d="M 10.8097 17.4365 11.7302 17.2249 12.5112 16.9827 12.7484 16.9006 13.4914 16.5562 13.7092 16.4304 14.3979 15.9905 14.5875 15.8259 15.2119 15.297 15.8592 14.6634 15.9108 14.5979 16.3789 13.8178 16.4941 13.5945 16.8632 12.8645 17.1318 12.0917 17.2025 11.8508 17.3657 11.05 17.3896 10.7999 17.444 9.9826 17.3896 9.16692 17.3656 8.91725 17.2025 8.11638 17.1317 7.87554 16.8631 7.10269 16.7479 6.87948 16.3788 6.14946 15.9105 5.36926 15.8591 5.30397 15.2117 4.67022 15.0223 4.50584 14.3977 3.97685 13.7088 3.53682 13.4913 3.41117 12.748 3.06675 12.5111 2.98475 11.7396 2.68905 11.6574 2.67728 10.7269 2.51908 9.85651 2.54383 9.60593 2.5558 8.79179 2.63821 8.5479 2.69734 7.75189 2.88689 6.98884 3.18025 6.76078 3.28443 6.04389 3.67833 5.8394 3.8237 5.18019 4.30823 4.47392 4.91563 4.41656 4.97579 3.91111 5.70116 3.77523 5.91239 3.30745 6.59768 3.27654 6.6749 2.90742 7.54317 2.72947 8.39159 2.68193 8.63808 2.57319 9.44882 2.57319 9.69992 2.5732 10.5189 2.68197 11.3296 2.72952 11.5762 2.94704 12.3654 3.04039 12.5986 3.35841 13.3516 3.7753 14.0552 3.91129 14.2666 4.41878 14.9092 4.59214 15.091 5.18036 15.6594 5.38509 15.8049 6.04415 16.2893 6.76088 16.6831 6.98923 16.7874 7.75207 17.0807 7.99629 17.1399 8.79208 17.3293 9.60599 17.4117 9.85691 17.4237 10.7271 17.4484 z M 9.24636 13.4605 9.15753 13.4387 8.61539 13.2621 8.11248 12.9928 8.03145 12.9504 7.5856 12.5955 7.51736 12.5348 7.14285 12.1051 7.09086 12.0299 6.79837 11.5395 6.59685 11.0062 6.56435 10.9207 6.46269 10.3589 6.45164 10.2682 6.45162 9.69924 6.55327 9.13746 6.56432 9.04693 6.76584 8.51366 6.79832 8.42812 7.09079 7.93768 7.14276 7.86254 7.51729 7.43272 7.96315 7.0778 8.03135 7.01717 8.11233 6.97473 8.61527 6.7054 9.15744 6.52884 9.24621 6.50697 9.81157 6.43027 10.3805 6.45522 10.4717 6.45522 11.0281 6.58185 11.1169 6.60372 11.6412 6.82842 11.7222 6.87086 12.1984 7.18474 12.2666 7.24537 12.6786 7.63952 13.0139 8.1014 13.0659 8.17654 13.312 8.69018 13.3445 8.77573 13.4961 9.32502 13.5478 9.89215 13.5588 9.98268 13.5071 10.5516 13.4961 10.6423 13.3445 11.1916 13.312 11.2772 13.0659 11.7908 12.7307 12.2527 12.6787 12.3279 12.2667 12.722 12.1985 12.7827 11.7223 13.0965 11.6413 13.139 11.117 13.3637 11.0282 13.3856 10.4718 13.5122 9.90289 13.5372 9.81167 13.5372 z " style="fill: white; stroke: black; stroke-width: 0; fill-type: evenodd" slic3r:area="-0.0021" />
		<path d="M 58.4612 41.0687 59.3817 40.8579 60.1627 40.6157 60.3998 40.5336 61.1429 40.1884 61.3607 40.0626 62.0494 39.6227 62.239 39.4583 62.8634 38.9294 63.5107 38.2956 63.5623 38.2301 64.0304 37.45 64.1456 37.2267 64.5147 36.4967 64.7833 35.7239 64.8541 35.483 65.0172 34.6822 65.0411 34.4321 65.0955 33.6148 65.0411 32.7999 65.0171 32.5503 64.854 31.7486 64.7832 31.5077 64.5146 30.7349 64.3995 30.5116 64.0303 29.7816 63.562 29.0014 63.5106 28.9362 62.8632 28.3026 62.6738 28.138 62.0492 27.609 61.3603 27.1698 61.1428 27.0442 60.3995 26.6989 60.1627 26.6169 59.3911 26.3212 59.3089 26.3095 58.3784 26.1513 57.508 26.1762 57.2574 26.1882 56.4433 26.2706 56.1994 26.3295 55.4 [... cut ...]
I noticed that in the DLP / SLA Preview window (where we hit `Export SVG`), the whole file is displayed correctly, including the last layer. This indicates that this isn't an issue with the Slicing process itself, but with how the file is saved. 
When I try to load the saved SVG into [SVG Viewer](http://garyhodgson.github.io/slic3rsvgviewer/) it doesn't work. Only after I delete the incomplete path and add closing tags it is displayed - but then obviously lacking the last shape in the last layer.

Any ideas on what could cause this and how to fix it are appreciated.

N4rki, 2018-02-21:
Here is my source file (3MF): 
[Train_Engine.zip](https://github.com/alexrj/Slic3r/files/1744848/Train_Engine.zip)

Slicing to SVG through the CLI only version (slic3r.exe Train_Engine.3mf --export-svg --layer-height 0.2) produces another bug: The created svg has only empty 'dummy' layers, except for the very last layer, which is seems to be generated correctly. This happens for STL and 3MF alike. Here's an example: 
[Train_EngineCLI.3mf.svg.zip](https://github.com/alexrj/Slic3r/files/1744804/Train_EngineCLI.3mf.svg.zip).
I assume that's due to incomplete porting to C++ and would classify it as not really relevant at this point, as the CLI only version without Perl is barely used.

Slicing through the normal console version via `Slic3r-console.exe Train_Engine.3mf --export-svg --layer-height 0.2` uses as far as I am concerned the same algorithm as `File -> Slice to SVG` in the GUI.
This is because the output is not as stated in this issue, but resembles the output of #4310. So not the same problem, but still a problem :)
SVG output: [Train_Engine.zip](https://github.com/alexrj/Slic3r/files/1744812/Train_Engine.zip)

The way I understood it is that slicing to SVG via `File -> Slice to SVG` , as well as using Slic3r-console.exe is done in the perl code (namely print.pm). Whereas going via `Window -> DLP Projector -> OK -> Export SVG... slices it in SLAPrint.cpp`

What I want to achieve is slicing a multimaterial 3MF file to SVG while being able to distinguish different materials in the SVG. If that distinction is by the color of each shape in the svg or by some id doesn't really matter. What's important though is that Slic3r produces a complete SVG and that the positioning is correct. So to even start, I either have to fix #4311 or fix #4310, because atm neither does both right.

Can you give an indication on what could be easier to correct? Unfortunately I still couldn't find a good way to debug the C++ Code that's initiated by Perl. The eclipse (with EPIC plugin) debugger doesn't step into any C++ function calls.

lordofhyphens, 2018-02-25:
I'd much rather fix it in SLAPrint.cpp.

lordofhyphens, 2018-02-25:
There an unrelated bug in which if the loaded config is set to auto infill width (which is the default) the infill generation crashes Slic3r with a division by 0. I've hotfixed that (defaults min spacing in that situation to 0.3mm instead).

lordofhyphens, 2018-02-25:
So there's a UI deficiency here I think at minimum. In my testing it is generating the entire SVG just not when you think it would (aka the UI lets you do stuff).

lordofhyphens, 2018-02-25:
With some further experimentation, if you close Slic3r it finishes writing the file (probably because the buffers aren't being flushed).

lordofhyphens, 2018-02-25:
Actually closing the file handle was left out during write_svg().

I've added a fflush() and a fclose(). While I'd usually rather do the IO in a C++ iostreams manner, the fprintf() method used here is easier to see what is going on.

closed 2018-02-25 by lordofhyphens

N4rki, 2018-02-27:
Thanks @lordofhyphens! Seriously, the work you do for this community is exceptional. Kudos for helping everybody out!

commit 6e283f69 (2018-02-25): Flush the output buffer when writing svg files from PrintSLA.cpp.

Fixes #4311