slic3r/Slic3r #4323: Removing Post-Processing Script crasches Slicer

foreachthing, 2018-02-27:
### Version
_Version of Slic3r used goes here_

_Use `About->About Slic3r` for release versions._

_Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. *Only* use normal Slic3r version IDs._

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

TheThirdOne, 2018-04-03:
I am able to reproduce this.

Stack trace:
```
Signal SEGV at /home/benjamin/Documents/workspace/Slic3r/lib/Slic3r/GUI/OptionsGroup.pm line 381.
	Slic3r::GUI::ConfigOptionsGroup::get_option(Slic3r::GUI::ConfigOptionsGroup=HASH(0x6c61808), "post_process", 0) called at /home/benjamin/Documents/workspace/Slic3r/lib/Slic3r/GUI/OptionsGroup.pm line 411
	Slic3r::GUI::ConfigOptionsGroup::create_single_option_line(Slic3r::GUI::ConfigOptionsGroup=HASH(0x6c61808), "post_process", 0) called at /home/benjamin/Documents/workspace/Slic3r/lib/Slic3r/GUI/OptionsGroup.pm line 418
	Slic3r::GUI::ConfigOptionsGroup::append_single_option_line(Slic3r::GUI::ConfigOptionsGroup=HASH(0x6c61808), "post_process", 0) called at /home/benjamin/Documents/workspace/Slic3r/lib/Slic3r/GUI/Plater/OverrideSettingsPanel.pm line 192
	Slic3r::GUI::Plater::OverrideSettingsPanel::update_optgroup(Slic3r::GUI::Plater::OverrideSettingsPanel=HASH(0x635d6e8)) called at /home/benjamin/Documents/workspace/Slic3r/lib/Slic3r/GUI/Plater/OverrideSettingsPanel.pm line 133
	Slic3r::GUI::Plater::OverrideSettingsPanel::set_fixed_options(Slic3r::GUI::Plater::OverrideSettingsPanel=HASH(0x635d6e8), ARRAY(0x6d56bd8)) called at /home/benjamin/Documents/workspace/Slic3r/lib/Slic3r/GUI/Plater.pm line 643
	Slic3r::GUI::Plater::_on_select_preset(Slic3r::GUI::Plater=HASH(0x611fb00), "print") called at /home/benjamin/Documents/workspace/Slic3r/lib/Slic3r/GUI/Plater.pm line 826
	Slic3r::GUI::Plater::__ANON__[/home/benjamin/Documents/workspace/Slic3r/lib/Slic3r/GUI/Plater.pm:830]("post_process") called at /home/benjamin/Documents/workspace/Slic3r/lib/Slic3r/GUI/PresetEditor.pm line 153
	Slic3r::GUI::PresetEditor::__ANON__[/home/benjamin/Documents/workspace/Slic3r/lib/Slic3r/GUI/PresetEditor.pm:156]() called at /home/benjamin/Documents/workspace/Slic3r/lib/Slic3r/GUI.pm line 213
	Slic3r::GUI::__ANON__[/home/benjamin/Documents/workspace/Slic3r/lib/Slic3r/GUI.pm:215](Slic3r::GUI::MainFrame=HASH(0x61131d8), Wx::IdleEvent=SCALAR(0x666ab08)) called at ./slic3r.pl line 126
	eval {...} called at ./slic3r.pl line 126
```

With one of my machines this error also happens when adding a new character to an empty post processing script.

foreachthing, 2018-04-04:
Even if I change the script parameters, it crashes if (while changing!) the parameters seem wrong.
There should not be an event triggered while typing in this textbox.

gege2b, 2018-04-19:
I got the output on my win10 for this crash, it seems there is a little more details in it : 
see #4386 for the linux output if needed

```
Error in function ConfigBase__set at src\perlglue.cpp:182: Trying to set non-existing option
         at C:/Users/gbandini/Dropbox/#reprap/Slic3r/lib/Slic3r/GUI/Plater.pm line 406.
        Slic3r::GUI::Plater::__ANON__("post_process#0") called at C:/Users/gbandini/Dropbox/#reprap/Slic3r/lib/Slic3r/GUI/Plater/OverrideSettingsPanel.pm line 168
        Slic3r::GUI::Plater::OverrideSettingsPanel::__ANON__("post_process#0", "sdqsds") called at C:/Users/gbandini/Dropbox/#reprap/Slic3r/lib/Slic3r/GUI/OptionsGroup.pm line 258
        Slic3r::GUI::OptionsGroup::_on_change(Slic3r::GUI::ConfigOptionsGroup=HASH(0x115b9ad0), "post_process#0", "sdqsds") called at C:/Users/gbandini/Dropbox/#reprap/Slic3r/lib/Slic3r/GUI/OptionsGroup.pm line 475
        Slic3r::GUI::ConfigOptionsGroup::_on_change(Slic3r::GUI::ConfigOptionsGroup=HASH(0x115b9ad0), "post_process#0", "sdqsds") called at C:/Users/gbandini/Dropbox/#reprap/Slic3r/lib/Slic3r/GUI/OptionsGroup.pm line 162
        Slic3r::GUI::OptionsGroup::__ANON__("post_process#0", "sdqsds") called at C:/Users/gbandini/Dropbox/#reprap/Slic3r/lib/Slic3r/GUI/OptionsGroup/Field.pm line 36
        Slic3r::GUI::OptionsGroup::Field::_on_change(Slic3r::GUI::OptionsGroup::Field::TextCtrl=HASH(0x115c4a48), "post_process#0") called at C:/Users/gbandini/Dropbox/#reprap/Slic3r/lib/Slic3r/GUI/OptionsGroup/Field.pm line 192
        Slic3r::GUI::OptionsGroup::Field::TextCtrl::__ANON__(Slic3r::GUI::Plater::OverrideSettingsPanel=HASH(0x89856a0), Wx::CommandEvent=SCALAR(0x115b9698)) called at C:\Users\gbandini\Dropbox\#reprap\Slic3r\slic3r.pl line 126
        eval {...} called at C:\Users\gbandini\Dropbox\#reprap\Slic3r\slic3r.pl line 126
```

Duckle29, 2018-06-13:
Even less helpfull crash error for me

#Version 
1.3.1-dev-c058ae9-x86_64.AppImage

#OS
Ubuntu 17.10

#Error on start
```
/usr/lib/x86_64-linux-gnu/libproxy.so.1: symbol _ZTTNSt7__cxx1119basic_ostringstreamIcSt11char_traitsIcESaIcEEE, version GLIBCXX_3.4.21 not defined in file libstdc++.so.6 with link time reference
Failed to load module: /usr/lib/x86_64-linux-gnu/gio/modules/libgiolibproxy.so
```

#Error on crash
```
/tmp/.mount_Slic3rALBgbD/AppRun: line 11:  2208 Aborted                 (core dumped) LD_LIBRARY_PATH="$DIR/usr/lib" "${DIR}/usr/bin/perl-local" -I"${DIR}/usr/lib/local-lib/lib/perl5" "${DIR}/usr/bin/slic3r.pl" --gui "$@"
```

There is no other output

foreachthing, 2018-06-13:
I don't understand those wx-widgets at all. But it would help, if the _on-change_-event was removed from that textbox (and many others as well).

lordofhyphens, 2018-06-13:
@foreachthing so how does the config get updated, then?

Duckle29, 2018-06-13:
Maybe when a slice is started, or when you save it?

foreachthing, 2018-06-13:
@lordofhyphens you're telling me, that at _every_ change, the config gets saved? No wonder the screen keeps flickering while typing. Same for the start/end GCode?
Like @Duckle29 said, maybe save it when one actually clicks save.

lordofhyphens, 2018-06-13:
@foreachthing EVT_TEXT on those boxes fires _on_change().

For post-processing, we can and should build a separate widget that allows for chains of scripts to execute (script A runs, then script B, etc), which does its updates on EVT_KILL_FOCUS instead of EVT_TEXT.

If you wanted to play around with using EVT_KILL_FOCUS in general instead of what usually fires _on_change, take a look at https://github.com/slic3r/Slic3r/blob/master/lib/Slic3r/GUI/OptionsGroup/Field.pm

lordofhyphens, 2018-06-13:
@foreachthing Saving when someone clicks "save" will break the "modified/dirty" config expectations. Slic3r allows a user to modify a parameter and use it without having to save it first. 

The kill focus event probably would behave better for gcode, but I haven't studied when the event is expected to fire and if there are any race conditions that one would need to be aware of.

foreachthing, 2018-06-14:
@lordofhyphens OK, understood! I thought you saved back to the config file, but you save the content into memory for later use. In VB.net I'd "save" such a setting on "leave". "KILL_FOCUS" sounds a lot like "leave"...

I'll have a look at EVT_KILL_FOCUS.