slic3r/Slic3r #4324: Building with Visual Studio 2013 - Questions

N4rki, 2018-02-27:
### Version
1.2.9-1272-gdc9d7ed7

### Operating system type + version
Windows 10 Pro

### Description
I’m reluctant to post yet another issue on here, but at this point I’m stuck and see no other way forward. I finally decided to go the VS 2013 route, simply because for a beginner like me the best way to really understand what’s going on in Slic3r is to have a somewhat graphical debugger that supports Perl AND C++. 
When using the CLI only Version (no perl), gdbgui works great but not so much with the full version of Slic3r because gdbgui doesn’t support perl. I’m also able to put the Slic3r source in eclipse + EPIC plugin and can debug the perl portion there but the debugger is not able to step into any C++ code.

So I’m trying to follow these instructions https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-Windows#building-with-visual-studio-2013 but things get confusing for me rather early.
I assume others were able to use these instruction successfully so I’m sure this is because my lack of knowledge and understanding. So the purpose of this thread shall be to help people like me and make it foolproof by giving some more info here and there.

### Here are my issues:
#### 1.
`nmake test` only fails one test (copy.t) as it should. However `nmake install` comes with two potential issues:
First off, I get a ton of `Cannot find "Module::Build" in podpath: cannot find suitable replacement path, cannot resolve link`. Is that normal? I couldn’t find anything similar online that would help to fix it.
Secondly, only a small part of the installation is located at C:\wperl64d (two folders: ‘bin’, ‘html’). The rest gets put into C:\perl (three folders: ‘bin’, ‘lib’, ‘site’).

Did I miss something? 
I find this strange because this way the given directories in step 5 and 6 don’t even exist: `set PATH=%PATH%;c:\wperl64d\site\bin;c:\wperl64d\bin` and `c:\wperl64d\lib\CPAN\Config.pm`. 
Sure I can create them, but why add an empty directory to PATH...

Am I supposed to move the contents of  C:\perl into C:\wperl64d?

Here is my Makefile and the CLI output from `nmake install`:
[Makefile + CLI output nmake install.zip](https://github.com/alexrj/Slic3r/files/1763954/Makefile.CLI.output.nmake.install.zip)

#### 2.
It is not really clear to me when to use which perl installation. Is Strawberry perl even needed and if so, which Version? I went with 5.24.3.1 just to be nearest to the wperl archive
I assume that we use the cpan from wperl, because only then I get the question and MyConfig.pm is generated.
I installed cygwin which I didn't have yet, added the packages and edited and moved MyConfig.pm

Okay so now using `perl build.pl` doesn't fail at Math-Libm, but at ZipArchive.cpp.
[build.pl CLI output.txt](https://github.com/alexrj/Slic3r/files/1764371/build.pl.CLI.output.txt)
Last few lines: 
```
In file included from src\libslic3r\Zip\ZipArchive.cpp:1:0:
src\libslic3r\Zip\../../miniz/miniz.h: In function 'FILE* mz_freopen(const char*, const char*, FILE*)':
src\libslic3r\Zip\../../miniz/miniz.h:2861:50: error: 'freopen_s' was not declared in this scope
       if (freopen_s(&pFile, pPath, pMode, pStream))
                                                  ^
At global scope:
cc1plus.exe: warning: unrecognized command line option "-Wno-undefined-var-template"
error building xs.dll file from 'src\libslic3r\Zip\ZipArchive.cpp' at C:/sperl/perl/lib/ExtUtils/CBuilder/Platform/Windows.pm line 129.
FAIL
! Installing ./xs failed. See C:\wperl64d\cpanm\work\1519759327.6028\build.log for details. Retry with --force to force install it.
The XS/C++ code failed to compile, aborting
```
I tried it with sperl 5.24.3.1 and actually got the same results:
[sperl build.pl CLI output.txt](https://github.com/alexrj/Slic3r/files/1764383/sperl.build.pl.CLI.output.txt)

So yeah that's where I am right now.  Maybe @bubnikv can give me a few pointers? :)

lordofhyphens, 2018-03-02:
I wish I could help more :(

lordofhyphens, 2018-03-02:
Something that probably does function with gdbgui is "attach to process". Alternatively, the wrapper exec that we ship with Slic3r probably works on Windows (I haven't tried it yet).

lordofhyphens, 2018-03-02:
Here it is on my Linux system:

[image]

bubnikv, 2018-03-02:
You may try to use the prebuilt Perl for Windows, that we use for the
development of Slic3r Prusa Edition.
https://github.com/prusa3d/Slic3r/wiki/How-to-compile-Slic3r-Prusa-Edition-on-MS-Windows

I expect that this distro will work with the upstream Slic3r as well.

N4rki, 2018-03-06:
@lordofhyphens I would be so happy with this solution, unfortunaltely attaching to process doesn't seem to work with gdbgui on Windows. Even with a simple helloworld it can only stop the process but not load any symbols or sourcefiles. I posted the issue on the gdbgui github.

Still: Do you attach gdbgui to the Perl Interpreter process or the Slic3r process itself? Because Slic3r doesn't have a PID on Windows.
Furthermore, are symbols and and sourcefiles loaded automatically or do you browse them manually thorugh the "fetch source files" textbox on the lefthand side?
Just to know how I should expect it to behave. I might also finally change to Linux now :)

I tried running the prebuilt executables with gdbgui (and just gdb) but the issue is that although symbols are loaded successfuly, gdb can't seem to read them because the behavior is exactly as if it was compiled without debugging functionality. 

@bubnikv I followed the instructions and further advice I got over at Prusa3D. You really made it a ton easier with your all in one package! I got quite far with Prusa Slic3r, but the upstream version doesn't compile with wperl because boost isn't found. 
Anyway, are these instructions intended to put the whole of Slic3r into Visual Studio (including the gui) or just a limited Command Line version?

bubnikv, 2018-03-06:
> Anyway, are these instructions intended to put the whole of Slic3r into
Visual Studio (including the gui) or just a limited Command Line version?

I don't understand. With the Prusa3D fork, you may just run cmake to
generate the visual studio project & solution, then open the visual studio
project, compile, set break points and you may debug the C++ parts of the
GUI Slic3r.

N4rki, 2018-03-06:
So which one is the main function that starts the GUI? I had assumed it's the main function in slic3r.cpp in the project "slic3r". If I run that not much happens no matter if I pass arguments or not. Also, most of the code in slic3r.cpp is disabled though a #if 0 (...) #endif statement. If I remove that, I can't compile it anymore because of 122 errors. I explained it in more detail here: [https://github.com/prusa3d/Slic3r/issues/720](https://github.com/prusa3d/Slic3r/issues/720)

bubnikv, 2018-03-06:
> it in more detail here: prusa3d#720

To debug with visual studio, you start the perl.exe with the slic3r.pl as a
parameter.
If you run cmake against the current Prusa3D master, this will be
pre-configured for you, just press F5.

N4rki, 2018-03-06:
> To debug with visual studio, you start the perl.exe with the slic3r.pl as a
> parameter.
> If you run cmake against the current Prusa3D master, this will be
> pre-configured for you, just press F5.

I'm sorry I seem to not get what you mean because it made no difference. 

So I pulled the current master and compiled everything. I chose Visual Studio 12 as a generator for cmake, then built it within the Visual Studio IDE. Worked like a charm. So you are saying to now debug with VS I open slicer.sln, set my breakpoints where I want (I chose `Model Model::read_from_file`) and then run slic3r via `perl slic3r.pl` from the preconfigured cmdline?
If I do that Slic3r starts and works normally but none of my breakpoints get hit. I added an .stl object to the plater which - as far as I know - should get imported via `Model::read_from_file` but nothing happens within Visual Studio. 

Hitting F5 in Visual Studio only runs the very first line of code in the main function of the slic3r project:     `boost::nowide::args a(argc, argv);` It skips the rest because of `#if 0....`. It finishes after a splitsecond.
I can set breakpoints in that one line there and they behave normally. But all the rest of the code isn't even reached so I can't do much with it. 

What am I messing up?

bubnikv, 2018-03-06:
> What am I messing up?

Maybe we messed up :-)  I shall test it myself.

bubnikv, 2018-03-07:
@VojtechKral Vojtech, would you please look into our CMake Visual Studio
generator and give an advice? Thank you.

vojtechkral, 2018-03-07:
@N4rki I'm looking into it... My suspicion is this might be due to perl spawning additional child processes. I'll try to figure something out.

N4rki, 2018-03-07:
@lordofhyphens Where did you enable debugging functionality (i.e. symbols) in order to attach the `perl slic3r.pl` process to gdbgui? So far I am adding -g to CXXFLAGS  in the Cmakelists.txt in folder src. Is that sufficient or is there another spot? And is that Cmakelists even used when I compile with Strawberry perl and `perl build.pl` or is that completely separate?

lordofhyphens, 2018-07-09:
FYI Visual Studio 2013 is not supported, as it does not support C++11 core features such as constexpr. 
https://msdn.microsoft.com/en-us/library/hh567368.aspx

@N4rki cmake build env is completely separate. 

I build libslic3r with -g and then debug the perl exec itself. If you want to include that on the Perl environment, you want to add it to xs/Build.PL.

closed 2018-07-09 by lordofhyphens

lordofhyphens, 2018-07-10:
@N4rki I can only speak for myself, but thank you. Use my full name. 

Sorry for dropping the ball back in March.

N4rki, 2018-07-10:
No worries at all :) I figured out soon after that the cmake build was a dead end for me anyway. Yes, was planning to use you full name, as stated in your profile and on the Slic3r website.

I might come back to this project once my developing skills are a bit more honed and I can be of actual help. It sure was a fun experience getting involved.

lordofhyphens, 2018-07-10:
@N4rki if you want to actively hone them on something useful, take a look at my 
 cppgui branch at https://github.com/lordofhyphens/Slic3r/issues 

:)

N4rki, 2018-07-10:
Just took a peek, I could really see myself helping out. Will come back to this in a few weeks time. When I get my hands dirty, be prepared for a couple more questions coming you way ;)

Patlax, 2024-01-04:
@N4rki Congrats to your bachelor thesis! Since you said you managed to build slic3r: with which compiler were you able to build it? And did you build the core libslic3r for development purposes or did you use the prebuild files?

I would be really greatfull if you could give me some hints since I bit off more than I could chew when trying to build the core libslisc3r with CMake and/Or QMake including boost 1.84 in Qt (6.5).

N4rki, 2024-01-06:
Hi Patlax, thanks! As you know it's been a long time ago and I have almost no recollection of the process. However I wrote a full compile instruction back then so my research insitute could continue the work after my  departure. Just found that on an old backup harddrive and converted the word to markup so it can be posted here with proper formatting. It is in German though so use your tool of choice to translate it. 

<details>
  <summary>Compile Instructions</summary>

**Multimaterial Slicing Software „IPA Slic3er"**
# Kompilierungsanleitung

Zunächst ist die Anleitung auf [**https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-Windows**](https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-Windows) sorgfältig durchzulesen und zu verstehen. Diese Anleitung behandelt aber NICHT die im zweiten Teil beschriebene Methode mit Visual Studio.

Folgendes ist zusätzlich zu beachten:

### 1. Installieren von Perl

Bei Benutztung des custom Slic3er Strawberry Perl 5.24 builds muss nach installation zunächst der Perl Package Manager „CPAN" hinzugefügt werden. Dazu mit Windows Taste + X Admin Menü aufrufen, dann mit Taste A in die admin cmd gehen und mit _cpan App::cpanminus_ installation starten. Kommt Fehlermeldung „Had problems unarchiving. Please build manually" -> Verzeichnis unter \c\Users\\<kürzel\>\.cpan\sources\authors\id\M\MI\MIYAGAWA\App-cpanminus-1.7043.tar.gz manuell entpacken und per cmd mit _perl Makefile.PL_manuell bauen (siehe [https://github.com/alexrj/Slic3r/issues/4089](https://github.com/alexrj/Slic3r/issues/4089)). Dann _cpan App::cpanminus_ -Befehl erneut eingeben. Nun sollte als Rückgabe kommen, dass CPAN bereits installiert ist.

 Im Text steht zwar, dass die neuste Strawberry Perl Version (5.26) auch funktionieren sollte, hier ist aber zu beachten, dass die „dependency archives" aus Schritt 4 dann nichtmehr passen und manuell zusammengestellt und kompiliert werden müssten.

### 2. Installieren der C++ Boost Libraries (V.1.63.0)

Wichtig: Die zip-Datei boost\_1\_63\_0.7z selbst darf NICHT im Verzeichnis C:\dev\boost\_1\_63\_0 oder in übergeordneten Verzeichnissen liegen, da die Installation der Library sonst fehl schlägt!
 Außerdem: „Perl command prompt" ist in dem Fall, dass man wie hier einen Strawberry Perl build installiert hat ( und nicht z.B. Citrus Perl) das normale cmd! Um Fehler zu vermeiden, cmd wieder als Admin ausführen und die drei Befehle ausführen.

### 3. Clonen des Slic3er Github Repositories

Vorgehen wie in Beschreibung. Es ist zu beachten, dass es beim Kompilieren prinzipiell zu Fehlern kommen kann, wenn in den verwendeten Pfaden Leerzeichen vorhanden sind (Ordnernamen müssen evtl. angepasst werden). Explizit gilt das hier für das Verzeichnis, in welches Slic3er geklont wird.
 Das gleiche gilt dafür, wenn Verzeichnispfade zu lange sind oder Netzlaufwerke verwendet werden. Beides war hier bisher aber kein Problem.

### 4. Entpacken der „dependency archives"

Wie beschrieben soll die Datei „slic3r-perl-dependencies-5.24.0-win-seh-gcc6.3.0-x64.7z" ins Slic3er Verzeichnis entpackt werden.

### 5. Build-Prozess

Was in der Beschreibung nicht steht, ist dass bei Benutztung der Sl3cer spezifischen dependency archrives der Befehl _perl build.pl --gui_ NICHT nötig ist (siehe [https://github.com/alexrj/Slic3r/issues/3870](https://github.com/alexrj/Slic3r/issues/3870)).

## Quicktipps

- Prüfen, ob perl build erfolgreich installiert wurde: cmd -> _perl --v_
- Prüfen, ob gcc passende Version hat: cmd -> _gcc --version_
- CPAN-Modul manuell installieren: cmd -> _cpan \<Modul\>::\<Name\>_, zb. _cpan local::lib_
- CPAN-Modul deinstallieren: Skript nötig! In Verzeichnis [\\saturn20.ipa.stuttgart\ipa\_350\abt\353\Hiwi-Ordner\JCJ-Valentin-Busch\Bachelorarbeit\Github-Repo\](////saturn20.ipa.stuttgart/ipa_350/abt/353/Hiwi-Ordner/JCJ-Valentin-Busch/Bachelorarbeit/Github-Repo/) cmd aufrufen -> _perl uninstall\_perl\_module.pl \<modul\>_

## Troubleshooting

Bei fehlschlagender Kompilierung können folgende Schritte helfen:

- CPAN updaten mit _cpan App::cpanminus_
- Strawberry Perl Distribution reparieren oder neuinstallieren (Windows-Taste -> „Programme" tippen -> Programme hinzufügen oder entfernen)
- Sind Leerzeichen in den Verwendeten Pfaden? (z.B. in Hiwi-Ordnern mit \<kürzel\>\_\<vorname\>\_\<nachname\>)
- Sind die Pfade zu lange oder werden unzugängliche Netzwerklaufwerke verwendet?
- Ist die Boost Libary richtig installiert? (Siehe Schritt 2.)
- Wurde das richtige Dependecy Archive verwendet?

## Weitere Hinweise

### Kommandozeile

1. Schnelle Möglichkeiten Kommandozeile direkt in Verzeichnis aufzurufen:

- Shift + Rechtsklick in Ordner -> „Eingabeaufforderung öffnen"
- In Adresszeile „cmd" tippen -> Enter

Aber: Damit kann cmd NICHT als Administrator gestartet werden!

2. Kommandozeile als Administator in Netzwerklaufwerken aufzurufen oder dort hin zu navigieren schlägt fehl! („Netzwerklaufwerk nicht gefunden")
 (siehe https://support.microsoft.com/en-us/help/3035277/mapped-drives-are-not-available-from-an-elevated-prompt-when-uac-is-co)

</details>

Whether this guide still 100% maps to today's implementation of Slic3er I don't know. To @Patlax and anyone lese reading this, the guide can be considered public domain and anyone is welcome to add it to the official documentation (after making sure it still applies). All the best!