slic3r/Slic3r #4328: [Feature request] Speed for the first N layers

ArtemKuchin, 2018-03-01:
Currently one can set speed only for the first layers. It is for good adhesion to the bed.
However, when printing at high speed plastic stretched a lot and those forces curl model up on the edges very strongly. If the first N layer were printed slower they would not contain so much stress, be flatter and stonger and resist curling up from the upper part of printer object.
For example, printing a long rectangular parallelepiped 10mm high. First 5mm can be printed slow, the fast. This will allow it to more likely stay flat on the bed.
I guess it is easy to implement.

ArtemKuchin, 2018-03-02:
Umm. received a notify from alexrj

> This might be something for you:
> https://github.com/alexrj/Slic3r/wiki/Conditional-Gcode-Syntax-Spec

Well, but how to modify move speed  and extrusion speed for each and every G code command based on  that? I just do not see how.

Or did you delete your comment already?

foreachthing, 2018-03-02:
That was me ... too early! That's why I deleted the comment as soon as I noticed. Sorry about that.

I _thought_ one could adjust the speed as well. But that's more complicated than just slow the motor down.

ArtemKuchin, 2018-03-14:
Of course, it can be done using the modifier. But, well, setting speed for the first layer can also be done by modifers :)