slic3r/Slic3r #4332: Progress bar while generating GCODE is constant

bzamecnik, 2018-03-03:
### Version

1.3.0-dev (unfortunately no commit in the About dialog). Date of files 2017-04-15. I can't test it on the latest, since download server doesn't work (#4329).

### Operating system type + version
OSX Yosemite 10.10.5.

### Behavior
When I generate GCODE of a larger model (101k facets, 1429 ccm^3, STL ~5 MB, GCODE > 370 MB) the progress bar at the bottom of the page always shows around 90%, not the real progress.

- it would be good at least to show the correct progress
- ideally we could also show elapsed time and estimated remaining time
- ideally there should be a button to cancel the operation

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

Unfortunately it's a file printed for a customer and cannot be published.

bzamecnik, 2018-03-03:
Update: I managed to download the latest OSX build, so I'll test it there.

Keavon, 2018-03-07:
I'm at over 50 hours now waiting for this to export and I also see the progress bar stuck at ~90%. Some smaller (but still relatively large) models I have exported took overnight to export, but this is even larger and taking days. I wish the progress bar could tell me how far along it is. I also feel like it should definitely not take this long.