slic3r/Slic3r #4335: item brims are not sequentially printed

tedder, 2018-03-05:
### Version
1.3.0 (git 19ea7a5), which is roughly 2018/01/17

### Operating system type + version
MacOS

### Behavior
With sequential printing and 'brim width' enabled, the print order is:
- print all brims
- print item A
- print item B
- ...

The brims should really be printed with the items.

#### STL/Config (.ZIP) where problem occurs
n/a

lordofhyphens, 2018-04-07:
This is probably an artifact of how the slicing is done; the  brim is calculated as a single piece in order to handle the condition of brim A running into brim B.

lordofhyphens, 2018-04-07:
Could you still provide a plate + config for the issue?

tedder, 2018-04-08:
Zip contains stl, gcode, and the config bundle. It doesn't show the sequential printing in the preview, which is why I included the gcode.

[sample.zip](https://github.com/slic3r/Slic3r/files/1887706/sample.zip)