slic3r/Slic3r #4336: Build fails on Raspbian at Time-HiRes

JonathanHolvey, 2018-03-05:
### Version
1.2.9 (Git 65a23b)

### Operating system type + version
Raspbian 8 (Jessie) Octopi 0.14

### Behavior
While following build instructions from [Slic3r plugin for OctoPrint](https://plugins.octoprint.org/plugins/slic3r/), running `sudo perl Build.PL --verbose` fails while building Time-HiRes-1.9754:

```
...
--> Working on Time::HiRes
Fetching http://www.cpan.org/authors/id/J/JH/JHI/Time-HiRes-1.9754.tar.gz ... OK
Configuring Time-HiRes-1.9754 ... FAIL
! Timed out (> 60s). Use --verbose to retry.
! Configure failed for Time-HiRes-1.9754. See /root/.cpanm/work/1520227993.988/build.log for details.
```

The log file shows a little more information, but I've never worked with Perl and I don't know where to start debugging:

```
$ tail /root/.cpanm/work/1520234788.2186/build.log
Looking for clock_getres()... found.
Looking for clock_nanosleep()... found.
Looking for clock()... found.
Looking for working futimens()... found.
Looking for working utimensat()... found.
You seem to have subsecond timestamp setting.
Looking for stat() subsecond timestamps...
Trying struct stat st_atimespec.tv_nsec...-> FAIL Timed out (> 60s). Use --verbose to retry.
-> N/A
-> FAIL Configure failed for Time-HiRes-1.9754. See /root/.cpanm/work/1520234788.2186/build.log for details.
```

lordofhyphens, 2018-03-25:
Unfortunately Time-HiRes isn't our code. :(

JonathanHolvey, 2018-03-25:
I thought as much. I asked for advice on Stack Overflow, and did make some progress...  
https://stackoverflow.com/questions/49373884/how-to-debug-program-compilation-when-perl-module-fails

closed 2018-03-25 by JonathanHolvey

lordofhyphens, 2018-03-25:
I could suggest using the C++ makefile, but it is currently very limited compared to the perl implementation (for the UI and some of the slicing toolpath)