slic3r/Slic3r #4337: Changing model orientation?

BlaiddDrwg82, 2018-03-06:
### Version 
_0.9.9_

_Use `About->About Slic3r` for release versions._

_Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. *Only* use normal Slic3r version IDs._

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_iOS X EL Capitan version 10.11.6_

### Behavior
* _I want to change the orientation of a model, so that the model lays horizontal on the bed. The model is of a bird skull and currently slices, tip of beak first and works its way up to the rest of the skull. Needs a ton of support material and super long print time. 
 _
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

Not sure

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_
[KOOKABURRA_Remeshed_skull 16_ 1_001.stl.zip](https://github.com/alexrj/Slic3r/files/1785562/KOOKABURRA_Remeshed_skull.16_.1_001.stl.zip)

JonathanHolvey, 2018-03-06:
You can achieve this in the current stable version of Slic3r. [From version 1.2.9](http://slic3r.org/blog/new-stable-1.2.9) you can right click the model in the 3D platter view and choose the X or Y axis from the **Rotate** submenu.

[image]

lordofhyphens, 2018-03-25:
This is either already implemented under "rotate/flip" or is a dupe of #3047. Closing.

closed 2018-03-25 by lordofhyphens