slic3r/Slic3r #4338: Placeholder [current_extruder] in startup

halentech, 2018-03-08:
### Version
1.3.0-dev
Slic3r-master.2018.02.26.1627.ce6b7c8.32bit

### Operating system type + version
Windows 10 x64

### Behavior
During startup of new printout (Start G-code section) is impossible do determine which extruder will be used for printing. In placeholder  [current_extruder]  is stored unusable value from previous print. There should be value which extruder will be used for print.

Iam using dual parking extruder and I need to park and choose right extruder for printing.
See: https://www.thingiverse.com/thing:2815697

Jan Jirousek

nwneisen, 2019-09-04:
@halentech I started taking a look at this issue for you but I haven't been able to get a value that looked like an unusable value, only zero based extruder indices.  Could you provide and example of the unusable value and the process you are using so I can reproduce them myself?

I did find that that current_extruder was not available in the start_gcode section.  There was already an issue for this, #4637, which I went ahead and created a PR for.

halentech, 2019-09-04:
Unusable value - I mean that in the value I expecting extruder number used for actual printing. But there is stored value from previous printing. For example, If I start printing with second extruder only and previous used extruder was first, in this variable  [current_extruder] is first extruder and not second.

halentech, 2019-09-04:
So printout is printed by wrong extruder and also coordinates are used for wrong extruder.

nwneisen, 2019-09-05:
Would you be able to provide the start_gcode you are using?

If you look at #4637 there is an example for switching extruders.  Specifically,
`{if [current_extruder] == 0}G1 X1.6 F4800 ;Extruder 0 location`
`{if [current_extruder] == 1}G1 X83 F4800 ;Extruder 1 location`
The first extruder will contain a value of 0 and the second will be a value of 1.  My testing of doing individual complete prints with each extruder has matched this example.

Also, are you using sequential printing with two models or is this two completely seperate prints with one model each?

Just trying to wrap my head around what I'm missing to reproduce your issue.  Thanks!