slic3r/Slic3r #4339: The output Gcode stays the same with different density inputs

powertide, 2018-03-09:
I'm using Slic3r by the source code I recently downloaded from github.
I tried to generate Gcode for rectilinear infill giving Slic3r different density inputs, 10%,20%,40%...100%, 
to see what the relationship between the infill density and path spacing is.
The output Gcode stays exactly the same until it reaches 100%. 
Should I change the settings to avoid this or is this a real bug?

Here is the configure file : with infill density=10%.

[config.txt](https://github.com/alexrj/Slic3r/files/1795939/config.txt)

Here is the stl file I used, though I don't really think it matters.
[Box(stl).docx](https://github.com/alexrj/Slic3r/files/1795954/Box.stl.docx)

foreachthing, 2018-03-09:
1. set 'Solid infill every' to 0 (zero). ('Print Settings' -> 'Infill' -> 'Solid infill every')
2. done.

powertide, 2018-03-12:
@foreachthing Thanks a lot for your help! I think I understand better about this "solid infill every" setting now.
After doing some tests with different densities, I have got some results which I can't explain.
I read from the Slic3r maual that the spacing = extrusion_width - layer_height * (1 - PI/4).
But I think it ought to be : spacing = (extrusion_width - layer_height * (1 - PI/4))*fill_density.
And the results of my tests validated my guess.
Don't know why the “fill_density” was left out on purpose @lordofhyphens .

Also, when the stl file changes into something more complicated, the spacing seems to be changing unexpectedly. For example, below is one layer from the slicing of a tooth model. How come the infill density changes in a layer while my config file doesn't suggest it? I'm using the same config file from "config.txt" above, except that "solid_infill_every_layers = 0" part of course, thanks to @foreachthing .
[image]

powertide, 2018-03-12:
I think it has something to do with the "solid infill threshold area“ setting, so I changed it to zero. But it didn't turn out as I thought. How do I get the same infill density throughtout the whole slicing without solid infill here and there ?

foreachthing, 2018-03-12:
I wasn't able to use your attached stl ... but try a Slic3r version from here: https://dl.slic3r.org/dev/
I wanted to try that with Benchy, since it sliced it similar to what you have only now it doesn't.
Maybe upload your stl directly (or zipped)?

powertide, 2018-03-12:
Sure. Below is the tooth stl I used.
@foreachthing 
[teeth.zip](https://github.com/alexrj/Slic3r/files/1801637/teeth.zip)

foreachthing, 2018-03-12:
Yeah, I know what you mean!
This is odd/strange behavior. But my guess is, this has something to do with the layers below and how steep those are. Maybe adaptive slicing will give you better results?
BTW: Have you looked at what Cura does? It's very similar...

powertide, 2018-03-12:
Unfortunately I need my layer height to be a fixed value so adaptive slicing won't be an option for me.
Thanks all the same.:-)

closed 2018-03-14 by powertide