slic3r/Slic3r #4342: Top layer ends over nothing

ArtemKuchin, 2018-03-14:
Version
Slic3r-master.2018.02.24.1611.ab89469.64bit

I've seen it many times but finely got annoyed by it to report.
In many cases i see top layer end over nothing. To at least solve the problem i used 3-4 top layers, but this is does not always work nice if i need good inside space.

It does not really matter if i used 5% infill or 75%. There are always gaps and top layer end in those gaps if it does not extend up to the external perimeter. For example, if something grows higher from that top layer.

IMHO, it should first draw a line then then start drawing top layer attaching it to that line. Always draw a border around top layer before drawing top layer.

[config.zip](https://github.com/alexrj/Slic3r/files/1812711/config.zip)
[image]

[vedro_gcode.zip](https://github.com/alexrj/Slic3r/files/1812713/vedro_gcode.zip)
[vedro_stl.zip](https://github.com/alexrj/Slic3r/files/1812715/vedro_stl.zip)

foreachthing, 2018-03-14:
You could use a _[modifier](http://slic3r.org/blog/modifier-meshes)_ in this part to use more infill where you need it.
Very easy to draw up a volume to use a modifier with.

ArtemKuchin, 2018-03-15:
modifier infill will fall into empty space, so, modifier infill must be many layers thick to support top layers
also, my example is very simple, in many cases the form is very complex and creating tons of modifiers
is just too difficult
Basically, many problems with slicer algorithm can be solved by manually applying modifiers in places where needed, but I think this particular problem is a slicer algo problem. Such thing should not just happen.