slic3r/Slic3r #4344: Printing outer walls appear to have gaps between layers

djoveryde, 2018-03-16:
### Version
1.3.0 - feb 26th 2018

### Operating system type + version
Windows 10 - Printing to Octoprint

### Behavior
When printing the outer walls they dont seem to connect to each other or to the infill. The initial layer however prints fine, but every other layer has a gap. I have turned thin walls off and same problem.
Running the same file through software called "3D Create and Print" prints out perfectly so issue is down to either a slicer setting or something else.
Same problem exists in 1.2.9 version of Slic3r.
https://www.facebook.com/photo.php?fbid=10155474657759639&set=gm.1760521373968684&type=3&ifg=1

#### STL/Config (.ZIP) where problem occurs
Any STL and config can be found here https://pastebin.com/kxbPsU26

ArtemKuchin, 2018-03-17:
The photo is so bad i cannot even understand  where the problem on it that you are reporting.
Please, make a good photo and draw an arrow where to look at.
Also, are you sure you are not underextruding? Actual filament size might be less. Maybe extruder steps are off.

djoveryde, 2018-03-17:
[image]
Infillls perfect, first layer perfect. But the wall itself does the 3 layers seperately from each other.
Any other slicer prints perfectly fine

ArtemKuchin, 2018-03-18:
I see it on one model today on my print and i KNOW that it is underextrusion. you will not see it on infill because it is usually sparse and you will not see it on 1st layers if it is too long. You should use digital caliper and print 1 shell box and measure actual shell width and also measure actually 1st layer height.

djoveryde, 2018-03-21:
Default Extrusion width was switched back to auto but the issue still exists. If I use the smaller 3D Create and Print software to make the gcode file it prints everything perfectly. I've even checked the extrusion setup and made sure steps are correct  to no avail. I will try a smaller filament diameter but I wouldn't understand why other slicers would work but slic3r itself doesnt

xoan, 2018-03-22:
Please, try to not post photos of the actual print in your machine, but the screenshot from the preview tab in Slic3r.

ArtemKuchin, 2018-03-22:
djoveryde, you will not move and will not solve anything until you get yourself digital calipers with at least 0.02mm accuracy, print a test cube and measure the real width of the shell. There are many reasons for such problems at each part of printer. I have no idea what 3D Create and Print software is and what settings it uses and how it uses them.

lordofhyphens, 2018-04-07:
@djoveryde FYI you can upload small ZIP files to Github by either drag-and-drop or the "selecting them" link below the editor. 

[image]

[4344.zip](https://github.com/slic3r/Slic3r/files/1885982/4344.zip)