slic3r/Slic3r #4345: Honeycomb infill idnore "Combine infill every [1] layers."

AndKe, 2018-03-17:
### Version
1.3.0dev (26feb)

### Operating system type + version
Ubuntu 17.10

### Behavior
with: "Combine infill every [1] layers."
Any type infill is printed on every Z level, as expected.
Except Honeycomb,  It will print it's perimeters every third layer, maybe with insufficient filament, so it's mostly gone - and barely visible or not existing in reality.

[image]

lordofhyphens, 2018-04-01:
Was pretty sure I fixed this for regular honeycomb months ago.

Also: The preview lies a bit about the actual flow. Could you provide a photo as well of a print where this is going on?

lordofhyphens, 2018-04-01:
@bubnikv have you noticed this in the Prusa3d fork? I remember having to boost the flow when this was all done in Perl. 

#3410 #1783

AndKe, 2018-04-01:
@lordofhyphens too late for photo, I threw those prints away long time ago.

The issue is very real , I discovered it "in reverse" :  First saw some prints with top layer being surprisingly less supported /dense on edges , did not mind it that much, then I dropped by the workshop as the printer printed one with "bad" infill, I stopped it, thinking something was wrong - then discovered that only some walls of infill had this issue, then verified using OctoPi Gcode viewer, and lastly using Slic3r preview. - which I posted.

lordofhyphens, 2018-04-02:
@AndKe I think I've got it fixed now, could you give it a try in the next build?

lordofhyphens, 2018-04-02:
[image]

closed 2018-04-02 by lordofhyphens

AndKe, 2018-04-02:
@lordofhyphens I downloaded the current dev build with timestamp 2 April , 04:22
Same issue remain, it can also can be seen in the screenshot of preview that you posted 4 hour ago.
-dev builds should have some git /build tag info in about screen.

lordofhyphens, 2018-04-02:
The "missing" layers on the edges are an artifact of honeycomb, not infill every N.

lordofhyphens, 2018-04-02:
[image]
Here it is at N=1. Better to open a separate issue for it.

lordofhyphens, 2018-04-02:
The download has the git commit in it by the way; 4db029c is the commit with the change for this issue in it.

lordofhyphens, 2018-04-02:
#4359 for reference. I'm locking this now.

commit 4db029c7 (2018-04-02): Use the surface thickness if it is available for internal infill, instead of passing the layer height for the flow.

Fixes #4345