slic3r/Slic3r #4346: download server - client ip blacklist

rob-miller, 2018-03-19:
I find that I get '403 Forbidden' errors for any connection to dl.slic3r.org when attempting t access through a specific ISP and not using a VPN.   I am in Tanzania.  According to the blacklist check at whatsmyip.com, the ISP / IP (41.212.67.254) that does not work is only on one blacklist - for not having a forward-reverse-confirmed DNS lookup, kind of a high bar - while the ISP that does work is on several blacklists.  

I understand why IPs get blacklisted, but it would be useful to note on the e.g. http://slic3r.org/download page to try using a VPN or similar if there are problems downloading.

lordofhyphens, 2018-03-25:
@alexrj please look into this.

rob-miller, 2018-04-10:
Behavior (and my ip) unchanged as of 10 April 2018.  Works fine if I VPN somewhere else.

ScottyDS, 2019-01-05:
I to am having the same "Forbidden" issue to the download page.
I have tried from my home IPaddress, as well as the work IP address, and get the same error.
I am in South Africa. My home IP address (DHCP address) is only blacklisted on one site, and the work address is not blacklisted anywhere, yet access to the download files are forbidden from both locations.

IPaddress my one IP address is 41.48.54.94. I dont have a VPN service, so sadly I have had to move on to using Cura.

This access forbidden problem is hurting Slic3r's reputation.

lordofhyphens, 2019-01-05:
@alranel is the person who owns the web space. I don't have any control over it.

@ScottyDS for releases there will be versions at github as well (I have updated the website page at slic3r.org)

Https://github.com/slic3r/Slic3r/releases/latest

lordofhyphens, 2019-01-05:
@rob-miller I've updated the download page to include a link to the GitHub releases page as an alternate.

closed 2019-01-05 by lordofhyphens

alranel, 2019-01-06:
Actually there are no blacklist involved. I had manually blacklisted some 41.* subnets because they were DoSing the download server. I re-enabled them now.