slic3r/Slic3r #4348: Missing Perimeter and/or Layers

DanyboyInSwitzerland, 2018-03-23:
### Version
Version 1.39.1-prusa3d-win64

### Operating system type + version
Win10 Pro, Version 1709, build 16299.309

### Behavior
*When loading a statue (downloaded from thingiverse), in the 3d-view it seems ok. After slicing, there is the perimeter missing in two spots, and on one spot, there's not even infill printed. On that non-printed spot, the gcode seems to act as if he was printing two individual objects, sitting on top of each other.
The phenomenon only appears on the main-body, the "extremities" and supports appear not affected.
From the preview, it really appears that layers are missing, as the curvutare extends over the gap.
Repairing the stl-file did not change the behaviour.
When checking the file in Fusion360, I cannot see any problem with the stl itself as the gaps do not appear there.
I tried with using different types and amounts of infill, as well as layer-heights and number of perimeters, but no help... :(
If this is an error from my side - PLEASE let me know, I am eager to learn!
* Just load the stl and slice.
* Obviously, those gaps should not exist.
* In the gap where nothing gets printed, the same layout gets printed as if it would lay support on the print-bed.

screenshot from the 3d-view:
[image]
screenshot from the preview-view:
[image]

_Is this a new feature request?_
I did not find anything similar looking through six pages of reports/requests.

#### STL/Config (.ZIP) where problem occurs
config used for slicing:
[HarleyQuinn_Config.zip](https://github.com/alexrj/Slic3r/files/1841454/HarleyQuinn_Config.zip)
stl used to slice:
[HarleyQuinn_stl.zip](https://github.com/alexrj/Slic3r/files/1841462/HarleyQuinn_stl.zip)

FidelCapo, 2018-03-23:
@DanyboyInSwitzerland Thanks for the report. If you are using Prusa Edition Slic3r, please write reports to our issues: https://github.com/prusa3d/Slic3r/issues
I tested your model and there are a lot of errors. In the current master branch (version 1.40.0), the model slice correctly. As a temporary workaround I attach a repaired model.
Models can be repaired, for example, by Netfabb or Meshmixer services.

[aa-repaired.stl.zip](https://github.com/alexrj/Slic3r/files/1841946/aa-repaired.stl.zip)

DanyboyInSwitzerland, 2018-03-23:
Hello Hurka,

 

Thank you very much for the quick reply!

 

Ok, so the problem was a combination of an older version of slicer due to customization by prusa, as well as a model with lots of errors that needed more that autorepair could handle.

So next time for a similar situation, I will use the more sophisticated services if autorepair cannot provide a satisfying result.

 

THANK YOU VERY MUCH FOR THE HELP!

 

Sincerely,

Daniel

 

Gesendet: Freitag, 23. März 2018 um 15:47 Uhr

FidelCapo, 2018-03-23:
@DanyboyInSwitzerland The problem isn't in Prusa Edition, but in this version (1.39.1) we have modified the auto repair and unfortunately did not work the way we expected. I'm currently testing a lot of auto-repairing on the variously damaged models, and it looks very good in version 1.40.0.

Lots of happy prints
Filip

DanyboyInSwitzerland, 2018-03-23:
Cool! Looking forward to the next update then...! :)
...and please keep up the good work!I am very impressed with slicer, this was the first time I had a real problem - and my printer is almost working non-stop since I received it two months ago... ;)
Cheers,Daniel

closed 2018-03-25 by lordofhyphens