slic3r/Slic3r #4349: Missing thin walled structures

M3Man03, 2018-03-26:
### Version
1.3.0-dev
### Operating system type + version
Win10 x64

### Behavior
On many .stl files I've tried, Slic3r seems to miss thin walled structures while they work in other applications like Simplify3d.  I have seen this on other models as well, not just this one.  It is almost like the sensitivity of Slic3r isn't as good as others when it comes to detecting lines.  Is there any way to adjust this?

Here's one example:

[image]
[image]
[image]

#### STL/Config (.ZIP) where problem occurs

[Slic3r_config.zip](https://github.com/alexrj/Slic3r/files/1848596/Slic3r_config.zip)

bubnikv, 2018-03-26:
Please attach the STL.

M3Man03, 2018-03-26:
The STL is proprietary and I won't attach it here.  I can see if I can make a small slice of it that produces the same results and I'll attach that.

foreachthing, 2018-03-27:
Question: How thick are the walls on the STL and how thick does S3D slice them?

M3Man03, 2018-03-27:
The walls are 0.30mm thick and S3D makes them 1 wall thickness which is set to 0.40, which is the diameter of my tip.  I've tried setting Slic3r to 0.25 wall thickness and tip setting, and other variations, but the walls never show up.  

It's almost like if the walls are under 0.40, Slic3r simply ignores them.

xoan, 2018-03-27:
Not in nozzle diameter, but in Print Settings → Advanced → Extrusion width (Perimeters)

PS. Anyway, if the wall in the model is 0.30, and S3D makes them 0.40, and Slic3r just ignore them (because of your config), it is not an Slic3r fault.

M3Man03, 2018-03-27:
That setting is at 100%.  Do I need to make that 0.30 in order for Slic3r to find them?

M3Man03, 2018-03-27:
Changing that setting had no effect.  I tried 0.30, 0.25, nothing caused the interior structure to appear.

lordofhyphens, 2018-04-01:
Slic3r (correctly) does not attempt to print structures smaller than your hotend's nozzle. Lying to the slicer about the nozzle diameter is not a solution either. 

Upload a 3D model you see the problem with and the paired exported config you see the issue with.

Also, from your posted config, change the "100%" and "100%" to "auto" or "default" and try again.

% widths are computed _over layer height_.