slic3r/Slic3r #4350: Feature Request: Custom Seam Position

daragill, 2018-03-29:
Hi there!

Slic3er seam options have been getting better over the years and think it handles seam the best out of many slicers out there.

I wonder if its possible to add a custom seam position as an option, being able to define where a layer starts and ends and therefore where its seam is. Similar to 'Rear', but being able to choose where that goes. 

There are instances I can think of where being able to put a seam exactly where I want it would mean that seam is less noticeable.

dartrax, 2018-03-29:
Hi,
take a look at #4268.
You'll find that I try to implement that into SlicerPE, because with it's penalty approach it is easy to accomplish. I've tested it for a while and it works great, although there are some bugs I still need to fix. I may find the time this weekend to do this (no garantee...)

daragill, 2018-03-30:
Hey dartrax!

Thanks for the response! Wow, you have done amazing work to try make this happen. I'm actually using PE edition of Slic3r so its amazing that its easier to do on that. Fingers crossed it get implemented.

dartrax, 2018-06-22:
Thank you for your thumbs up, @scalablecory. Did you notice that you can actually download a compiled package for win 64bit to test that feature? You'll find the link with the [pull request](https://github.com/prusa3d/Slic3r/pull/728).

scalablecory, 2018-06-22:
@dartrax I did not. I'll give it a try and do a code review to try to speed this along...

Jochem22, 2021-06-04:
any idea if this will ever be added to slic3r?

daragill, 2021-06-04:
> any idea if this will ever be added to slic3r?

This is a feature now of PrusaSlic3r called ‘paint on seams’.

Jochem22, 2021-06-05:
> > any idea if this will ever be added to slic3r?
> 
> This is a feature now of PrusaSlic3r called ‘paint on seams’.

somehow slic3r gives me way better prints then Prusa or Superslicer so i was hoping it get added to slic3r ;)

supermerill, 2021-06-05:
> somehow slic3r gives me way better prints then Prusa or Superslicer so i was hoping it get added to slic3r ;)

if you a little print that prints better with slic3r, I'm interested in the object+config used to showcase that. To see if something can be improved/restored.

Jochem22, 2021-06-05:
> > somehow slic3r gives me way better prints then Prusa or Superslicer so i was hoping it get added to slic3r ;)
> 
> if you a little print that prints better with slic3r, I'm interested in the object+config used to showcase that. To see if something can be improved/restored.

Sure, how can i give you the info + pics?

supermerill, 2021-06-22:
as you wish, here or on a new issue on the susi git.