slic3r/Slic3r #4352: Incorrect colors for infill and outline in layers tab

xoan, 2018-03-31:
### Version
Current master: 1.2.9-1291-gce6b7c8f

### Operating system type + version
Arch Linux

### Behavior
I guess since merge PR for color scheme, some colors changed in «Default» scheme.

Infill is now black, and outline is red, like perimeters.

[image]

It seems that I have 3 perimeters configured, but actually, I have 2. The outer line is the outline of the layer part.

foreachthing, 2018-03-31:
I`ll have a look. I don`t think I changed any of the default colors... but something happened, I guess?!

xoan, 2018-03-31:
I've updated the prior message to attach the screenshot of the layers tab before the last commit.

xoan, 2018-03-31:
Also in 2D tab, when there are more than one copy of the object, the selected one have no different color.

[image]
*After last commit*

[image]
*Before last commit*

Samir55, 2018-03-31:
@foreachthing 
For the 2D tab. The instance_brush (the selected instance of the selected object) color shouldn't be the same color of the selected_brush. 
```Perl
 -    $self->{instance_brush}     = Wx::Brush->new(Wx::Colour->new(255,128,128), wxSOLID);
 -    $self->{selected_brush}     = Wx::Brush->new(Wx::Colour->new(255,166,128), wxSOLID);
 +    $self->{instance_brush}     = Wx::Brush->new(Wx::Colour->new(@BED_SELECTED), wxSOLID);
 +    $self->{selected_brush}     = Wx::Brush->new(Wx::Colour->new(@BED_SELECTED), wxSOLID);
```

lordofhyphens, 2018-04-01:
Fix has been merged, closing.

closed 2018-04-01 by lordofhyphens