slic3r/Slic3r #4353: Slic3r, master version alternates T0/T1 with extruder 2 selected.

rickyrockrat, 2018-03-31:
### Version
ce6b7c8fc229c2ed16c524ba4b0f1ad8e305557e

### Operating system type + version
Devuan - Jessie

### Behavior
Set all extruders to 2:
grep '_extruder ' /home/doug/.local/share/RepetierHost/slic3r_settings.ini 
infill_extruder = 2
perimeter_extruder = 2
solid_infill_extruder = 2
support_material_extruder = 2
support_material_interface_extruder = 2

* _Describe the problem_
I get alternating T0/T1 Gcode
* _Steps needed to reproduce the problem_
Set all extruders to 2, slice a STL.
  * _If this is a command-line slicing issue, include the options used_
From Repetier, it's command line is:
--load slic3r_settings.ini --dont-arrange -o composition.gcode composition.amf
* _Expected Results_
All T0 to be T1
* _Actual Results_
Alternating T1/T0
  * _Screenshots from __*Slic3r*__ preview are preferred_

Failing to figure out the code ( I can't figure out who calls the methods in GCodewriter), I wrote a work around:

[slic3r.txt](https://github.com/alexrj/Slic3r/files/1865923/slic3r.txt)

lordofhyphens, 2018-04-01:
[4353.zip](https://github.com/alexrj/Slic3r/files/1866685/4353.zip)

With this configuration, I don't see any T0 in the output.

lordofhyphens, 2018-04-01:
I'm closing for now (can't reproduce). Please provide a full export of the config (run Slic3r by itself, get the problem to happen, and hit Ctrl-E) as well as a sample model you see the problem on.

lordofhyphens, 2018-04-01:
I will reopen if you can reproduce.

closed 2018-04-01 by lordofhyphens

rickyrockrat, 2018-04-06:
Here is the configuration.  I'll verify again.
[slicer-alternating.txt](https://github.com/slic3r/Slic3r/files/1883167/slicer-alternating.txt)