slic3r/Slic3r #4358: 1.3.0 missing auto-arrange and focus on stl load.

AndKe, 2018-04-02:
### Version
1.3.0. -dev

### Operating system type + version
Ubuntu 17.10

### Behavior
in 1.2.9 , adding a stl, arranged the STL in the middle of the print area, and the 3D preview were showing the display.  (easy, smooth, nice)

in 1.3.0 we see the corner of the bed area, on loading an .stl, it's centered on the 0.0 coordinate, mostly outside the bed.
Then user may click "Arrange"  - which make the part disappear to the center of the bed.
Then user must zoom out/pan to see it all.

Why ? - I can't believe that is intentional ?

lordofhyphens, 2018-04-02:
Auto arrange on load was removed with a pull request a few months ago. Leave auto arrange on for x/y all the time if you want it there.

What could be done is recentering the camera when the arrange button is clicked.

AndKe, 2018-04-02:
>>Auto arrange on load was removed with a pull request a few months ago. 
Why would anyone do something like that ?

>>Leave auto arrange on for x/y all the time if you want it there.
I did not know there is a place that I could do that ?  - where ? - can't find

>>What could be done is recentering the camera when the arrange button is clicked.
Absolutely, not to mention bring back the old "auto arrange+autocenter camera" as it were.

This is a major regression in user experience:

before the experience was like;
load stl, (maybe place more)
print
delete stl
(repeat)

now it is like:
load stl, 
place it
move camera to verify placement /place more
print
delete stl

combined with the fact that Slic3r does not detect file system updates for loaded .stl file, it causes a lot more work.
- still - the choice to remove the old autoplace/move camera is probably not made by a sober person and need to be rethought...  :)

lordofhyphens, 2018-04-05:
@AndKe #4156 was the PR that did it. 

@GilesBathgate you should probably make this behavior configurable.

GilesBathgate, 2018-04-05:
@lordofhyphens I don't understand what you mean, the behaviour is configurable via the auto center option in the preferences. When the auto center option is on (default) the behaviour is the same as 1.2.9.

lordofhyphens, 2018-04-05:
@GilesBathgate @Andke wants the initial load to auto place (which is different than auto arrange all the time). 

I think I would prefer to have the arrange button recenter the camera view

foreachthing, 2018-04-05:
Please leave the camera where it is - it's _there_ for a reason. _I_ can adjust the camera myself, thank you. Or make it optional.

AndKe, 2018-04-05:
@GilesBathgate after "discovering" auto center, I am quite happy with it as-is.

It should be on by default, convenient for most users. (I cannot think if a scenario where I'd rather have it centered on 0,0 instead.)

GilesBathgate, 2018-04-05:
@lordofhyphens But I don't understand how this relates to my change. If you look at the commit:
1) I've moved the code that sets $need_arrange = 0; into the block of code that sets $need_arrange = 1; (which will not get executed with the option set on as in 1.2.9)
2) I've added $o->align_to_ground; # aligns object to Z = 0 and $o->add_instance(); 
(neither of these lines will get executed with the option set on as in 1.2.9)

I am still trying to understand why my code changes are a regression.

GilesBathgate, 2018-04-05:
@AndKe Well the auto center option in the preferences is on by default. 

**lordofhyphens commented on 20 Oct 2017**
>I'm okay with changing this behavior slightly, especially as autoarrange is on by default.

lordofhyphens, 2018-04-05:
They weren't a regression *per se*, just the usual headache involving a UI change (as the initial plating did auto place in the center of the plate before).

If @Andke is satisfied now then I will close the issue.

AndKe, 2018-04-05:
yes, issue can be closed, but please consider making the auto-center be "on" by default.
That will be a much better first-time experience to new users / and those tho did not yet look for it in preferences.

GilesBathgate, 2018-04-05:
@AndKe It is the default https://github.com/slic3r/Slic3r/blob/master/lib/Slic3r/GUI.pm#L83 assuming 1 means true in perl!

AndKe, 2018-04-05:
I see, so - maybe me having a 1.2.9 config did not assert that ? - it probably only works on first run ?

in such case, it would be best to handle it by detecting if the config file were from ver <1.3.0, or if the config file had no setting for that,  and in such case, add it .. as true..

GilesBathgate, 2018-04-05:
@AndKe I think it warrants further investigation. I will try to find time to take a look.