slic3r/Slic3r #4359: Honeycomb Infill draws unsupported border lines.

lordofhyphens, 2018-04-02:
### Version
4db029c

### Behavior
Honeycomb infill bordering on the perimeters draws unsupported lines. This makes these areas have a weak binding to the rest of the infill and to the perimeters.

[image]

Suggested solution: Incorporate the snippets for the borders into the draw pattern for honeycomb.

#### STL/Config (.ZIP) where problem occurs
[model_and_config.zip](https://github.com/alexrj/Slic3r/files/1868090/model_and_config.zip)

lordofhyphens, 2018-04-02:
Confirming that this behavior is also present in the prusa3d fork, fyi @bubnikv 

@AndKe fyi

supermerill, 2018-04-03:
I was thinking it was made to remove many wipes moves and so to increase the speed of the infill.

AndKe, 2018-04-03:
it certainly does nothing good as-is.

xoan, 2018-04-03:
All the infill patterns do the same.

[image]

As @supermerill says, it removes a lot of wipe and retract moves, mainly in convex corners like the one above.

Don't remember when this was commited, but I think it was a huge improvement in speed and printability

@lordofhyphens what are the downside of it?

bubnikv, 2018-04-03:
>  All the infill patterns do the same.

Roughly, yes.

None of the infill types try to place the connecting lines (the pieces of
perimeter) one over the above. I don't think this is even feasible.

The grid infill has yet another issue, where the pieces of perimeter for
the two directions in a single layer may overlap, creating over extrusion.
If there is enough space inside the object, then this is usually not a
problem as the over extrusion is spread inside, but it may be issue for
narrow objects. The stars / triangles / cubic infill has the same issue.

There is some difference between the honeycomb and the
grid/stars/cubic/triangles infill in the way how the infill lines are
connected by the perimeter pieces. The grid/stars/cubic/triangles try to
connect the infill lines into a continuous sweep, while the honeycomb
infill creates loops. Naturally the first is better, and the honeycomb
creates loops by incidence: The honeycomb infill lines are trimmed by the
polygonal boundary using a clipper library, and the clipper library creates
these loops. The clipper library may be switched to not connect the infill
lines with the perimeter pieces at all, but the clipper is not able to
generate a continuous sweep by itself, and the continuous sweep would
require significant changes of the honeycomb infill algorithm.

Is not having the perimeter pieces supported an issue? With enough infill /
perimeter overlap these pieces are supposed to be sufficiently fused to the
inner perimeter.

xoan, 2018-04-03:
Totally agree.

lordofhyphens, 2018-04-03:
Doesn't the modified rectilinear create loops? I don't have the system in front of me.

lordofhyphens, 2018-04-03:
As an aside, I created the issue for @AndKe. 

I think that the current system is a decent tradeoff myself and some infill patterns are better than others for every N.