slic3r/Slic3r #4360: Spiral vase works fine on vertical objects, but gives compensation artefacts on slopes

Leozolt, 2018-04-02:
The error looks like the slicer makes a spiral on the Z layer (
[image]
) 

but compensates the **diameter** of the spiral on a single spot, instead of doing a conical spiral (
[image]
).

Using slic3r 1.3.0-dev but had the same issue before tho it's less visible with a thinner nozzle (0,4 .. 0,6mm).

Using 1,2mm nozzle in the print pictures also happens if there is *no seam* in the stl on the spot where it happens.

The error appears at the same spot where the spiral begins.

[image]
[image]

It's not visible in preview or layer preview as they only show layer circles an no actual path preview...
[image]

If a stl is required i prefer transmitting it on a non public space for this model.

Thank You!

lordofhyphens, 2018-04-02:
@Leozolt could you add a copy of the configuration? Use CTRL-E (or File->Export Config)

lordofhyphens, 2018-04-02:
Here's a possible test case @Leozolt can you confirm this model with your config causes the problem?

[possible_test.zip](https://github.com/slic3r/Slic3r/files/1869665/possible_test.zip)

Leozolt, 2018-04-03:
Printed out the slopevase.stl
[image]

The change in diameter per "layer" is clearly visible/feelable on the issue line but i am not able to photograph it accordingly.

Same issue with my config (needed to reduce to 15mm/s for the slopevase the diameter is too small) 
[config.zip](https://github.com/slic3r/Slic3r/files/1869773/config.zip)

Leozolt, 2018-04-03:
Is it possible to apply the "hack" on the x y plane to make the diameter grow continually?

bubnikv, 2018-04-03:
Maybe. One would have to find corresponding points between two successive contours and then to linearly  morph from one contour to the other. This is not too difficult to implement.

Other solution would be to calculate the spiral over the input surface, but that is much more complex to do.

Leozolt, 2018-04-05:
ok, it would require to calculate a spiral (conical helix segment) from one layer to the next and subdivide it to small linear segments. ??

mesheldrake, 2018-04-06:
No, not always a conical helix segment. Think of the general case - any old spiral vase type model - where it's not necessarily conical. That's where @bubnikv 's first suggestion makes more sense.