slic3r/Slic3r #4366: [doc] update build procedure for  openSUSE LEAP 15.0

kdupke, 2018-04-04:
### Version
Master as of 2018-04-04

### Operating system type + version
openSUSE LEAP 15.0

### Behavior
Installation prerequisites are:
libboost_headers1_66_0-devel libboost_thread1_66_0-devel libboost_system1_66_0-devel libboost_filesystem1_66_0-devel

Installation of needed libs can be done by command "zypper in libboost_headers1_66_0-devel libboost_thread1_66_0-devel libboost_system1_66_0-devel libboost_filesystem1_66_0-devel"

Build needs 
export BOOST_INCLUDEPATH=/usr/include/boost/
export BOOST_LIBRARYPATH=/usr/lib64/
before using perl Build.pl

_Is this a new feature request?_
Just update the doc page at
https://github.com/slic3r/Slic3r/wiki/Running-Slic3r-from-git-on-GNU-Linux

kdupke, 2018-04-04:
for Build.PL -- gui it needs:

package "wxWidgets-3_0-devel freeglut-devel libXmu-devel libXi-devel"

Install with "zypper in wxWidgets-3_0-devel freeglut-devel libXmu-devel libXi-devel"

lordofhyphens, 2018-04-05:
It is a wiki. Help yourself.

closed 2018-04-05 by lordofhyphens