slic3r/Slic3r #4367: Core Dump with openSUSE LEAP 15.0

kdupke, 2018-04-04:
### Version
master branch as of 2018-04-04
1.2.9-1296-g7b8369d0

### Operating system type + version
openSUSE LEAP 15.0

### Behavior
- Build slic3r and gui
- start ./slic3r.pl
- core dump

What is next?

kdupke, 2018-04-04:
Running ./Slic3r on the same system from pre-build
slic3r-1.3.0-dev-4db029c-linux-x64.tar.bz2
works fine.

lordofhyphens, 2018-04-05:
Where to go next is a very good question. I don't have any OpenSUSE systems to test compilation with.

You can try gdb `which perl` and then load the core file from there to get a backtrace. I would be curious to know if command line slicing works (and it is just the GUI).

kdupke, 2018-04-16:
Sorry for the delay.

Looks like this is an openGL issue. Can I build without openGL (2d only maybe) to verify?

Best regards,
kai

lordofhyphens, 2018-04-16:
You can build, but you'll need to comment out most of the 3D preview system if you want to use the GUI without opengl