slic3r/Slic3r #4368: UI suggestion: Show visual warning for provably "wrong" configuration settings.

lordofhyphens, 2018-04-05:
There are several advanced configuration settings that people still feel like they need to mess with and either misunderstand what the meaning of the config is or not know that they've gone past some hard limits in the printer. Then they blame Slic3r :( The primary offender for this is the Advanced width options.

How would putting up something like the classic ! in the row the configuration option work out for people? It'd be primarily to warn the user that something is probably not what they actually want. We wouldn't restrict users from doing silly things, but remind them that it is indeed silly.

The problem with restricting people from doing it (and providing an override) is that at some point users will get used to setting that override and then we're back where we started.

xoan, 2018-04-05:
It should be a nice improvement. Maybe by using an icon with a tooltip explaining the warning.

[image]

gege2b, 2018-04-19:
Showing people they're doing something that may be bad is the only thing I can imagine here

I agree with @xoan 's suggestion, and maybe also add a little comment on top of the advanced settings to warn the user that they're going to change critical settings and they really should know the impact of such changes in the final result, and tell them that if a print comes out dirty, maybe they should try with default advanced settings

And maybe also add this kind of warning directly into the Gcode, right before the out-of-limit settings like 

```
; ! not optimal setting set by user
; perimeters extrusion width = 0.001mm 
 ````

this should make debuging of dirty prints more easy