slic3r/Slic3r #4370: Presure advance

Tinchus2009, 2018-04-05:
### Version
latest dev version

### Operating system type + version
_win10 64b

### Behavior
The option pesure advance, is suposed to be cpable of increasing pressure at the begginign of a movement and decrease presure at the end of it too, right?
Im not being able to make it work... I have used values starting from 0.1, 0.5 and I got till 2. And every print was exactly the same. In particular im looking to avoid those bulkie corners seen in box shapes for example. In other slicer like simplify or cura too, there is an option called costeing, this would be its equivalente right? Seems to not be working?

lordofhyphens, 2018-04-06:
Pressure advance is dark magic and experimental in Slic3r. Use autospeed instead, or if you are using Marlin, linear advance.

Tinchus2009, 2018-04-06:
yeah... autospeed... I tried it but it does not give the result Im looking for. Autospeed control the speed in order to have less speed changes across the print, presure advance was suposed to control the presure at the begining and end of a path.... it doesnt solv the common problem if these bulkie corners.

lordofhyphens, 2018-04-06:
Autospeed is supposed to keep constant pressure in the hotend.

timor, 2018-04-23:
Your bulky corners could also result from excessive 'ringing'.  In that case, lowering acceleration and/or playing with the jerk values might help, but the root cause would be a 'wobbly' mechanical setup of the printer.