slic3r/Slic3r #4374: UI - "Reload from disk" resets modifiers and parts added in settings

Oekn5w, 2018-04-10:
### Version
6454abfd5fbeb7140edcadee6a4b358e93fba35b

### Behavior
Reloading the selected object (feature implemented after feature request #1466) dismisses any parts and modifiers added via the settings dialog.

### Expected
Copying the parts and modifiers and their configuration to the reloaded object
Or (UI option?)
Also reloading the volumes of the parts and modifiers and copying their configuration

lordofhyphens, 2018-04-10:
So why is there an assumption that a "reload" of the STL shouldn't discard modifiers/settings? 

This is important to ask because in this case, changing the behavior is just going to lead to someone else expecting the other way.

foreachthing, 2018-04-11:
@lordofhyphens, well, true. But then I print a lots of parts from CAD/FEA and they are not always designed to be printed in best orientation. So, one needs to rotated or even split those parts prior printing. But then changes come along and you need to redo your rotation and splits again. This can be a very time consuming process.

And foremost, because it's the same STL - it's the very same part.

Picture this: You got a part where you need modifiers for better infill and strength. You set this up and it may take you 10 minutes to do so. Now you find out you need to change your model just a tiny bit (_any_ change for that matter). 
Well, you start all over! And maybe you'll be able to match your previous setup... maybe.

bubnikv, 2018-04-11:
Yet better would be to generate an annotated amf/3mf, where the modifier
meshes would be recognized as such by Slic3r during import from their
annotation produced by the CAD.

Oekn5w, 2018-04-11:
@lordofhyphens the description from from @foreachthing is basically what triggered the writing of this issue, updating the model and then having to redo the solid infil etc.
IMO the assumption to keep the changes when reloading is fair, as reload from file suggests an updated geometry, while maintaining everything else. To start new one could also add a new object at the inconvenience of having to select a file.

From the reload_from_disk function
https://github.com/slic3r/Slic3r/blob/c3cb01eb93dd9b078b627acbc77acf526efb4637/lib/Slic3r/GUI/Plater.pm#L2322-L2326
the changes made to the slicing options get copied if the number of volumes are the same (which they probably are in the amf/3mf files mentioned by @bubnikv, I can't test this with the CAD software I have). This condition is false when additional files are added via the settings dialog, discarding any changes made.

@foreachthing keep in mind, splitting diables reload from disk, see #4269

I'm working on this issue (this one, not #4269), although the changes are not yet ready to commit.

@bubnikv the goal would be that the user also has a straight-forward way if they are limited to STLs

foreachthing, 2018-04-19:
> but any mesh modifier should be wiped to avoid unwanted effects.

I couldn't disagree more! Maybe unwanted by you - but desired to me.
Mesh modifier **should** be kept and even be enabled (_well, then make it optional! Or ask the user if he/she wants to delete, keep or disable existing modifiers_). Just because I change a hole diameter at some place, I don't want to loose all other settings/modifiers at the other place of my part - it's the same part after all. And I can edit my modifiers myself. I just don't really want to redo 'em.

Oekn5w, 2018-04-19:
I actually have overhauled the reload function to ask the user how to handle added parts and modifiers (reload them too oder deleting all of them), although the testing of it is quite difficult, as my CAD only generates single-object, single-volume STLs

VanessaE, 2018-04-19:
Just reload the part and keep the modifiers, settings, original position (relative to the model's origin), etc.  Let the user decide after it's loaded if something needs tweaked.

lordofhyphens, 2018-04-24:
Closing as related PR is merged.

closed 2018-04-24 by lordofhyphens