slic3r/Slic3r #4375: Use _octoprint._tcp Zeroconf service

probonopd, 2018-04-10:
https://github.com/slic3r/Slic3r/blob/19ea7a596a19e7a59ddb2ae2f3dde8c107bd9d11/lib/Slic3r/GUI/PresetEditor.pm#L1326

seems to search for `_http._tcp` services when OctoPrint in fact [advertises itself](https://github.com/foosel/OctoPrint/wiki/Plugin:-Discovery#zeroconf-service-_octoprint_tcp) as `_octoprint._tcp`.

cc @foosel

probonopd, 2018-04-10:
Currently my instance of [OSMC](https://osmc.tv/) shows up as a 3D printer in Slic3r.

lordofhyphens, 2018-04-10:
Pull request instead?

lordofhyphens, 2018-04-10:
@foosel let me know if something is different than the spec and I'll update as necessary.

closed 2018-04-10 by lordofhyphens

probonopd, 2018-07-28:
Thanks, https://github.com/slic3r/Slic3r/commit/c3cb01eb93dd9b078b627acbc77acf526efb4637 addresses this. Works for me.

commit c3cb01eb (2018-04-10): look for octoprint as a octoprint service, not http.

Reference:
https://github.com/foosel/OctoPrint/wiki/Plugin:-Discovery#zeroconf-service-_octoprint_tcp

Fixes #4375