slic3r/Slic3r #4380: [Feature Request] Flag bridging moves in verbose GCode comments

timor, 2018-04-14:
I would like to be able to see in the generated GCode where bridge moves take place.  This could be used for post-processing scripts.  Or is there a different method of identifying those moves for post processing?

lordofhyphens, 2018-04-14:
@timor That should be sufficient, although I don't remember offhand if all paths that use bridging flow are marked is_bridge() or just what people usually think of as bridges.

closed 2018-04-14 by lordofhyphens

timor, 2018-04-15:
Thank you very much!
I will try and see if these are the perimeters that I expect.

lordofhyphens, 2018-04-16:
There's bridging infill too that will show up.

lordofhyphens, 2018-04-19:
I believe it's treated the same way (because to Slic3r there is no real difference, bridge flow is bridge flow).

commit 7457cfda (2018-04-14): Outputs (bridge) in verbose gcode if path extruded is marked as bridge. Fixes #4380