slic3r/Slic3r #4385: Every keypress in a text aera triggers a config validation and UI refresh

gege2b, 2018-04-17:
### Version
One of the latest 1.3.0-dev (downloaded approx. two weeks ago) but the problem is not new

### Operating system type + version
Windows 10

### Behavior
* _Describe the problem_
Eveytime a key is pressed when in a text area (like post processing or custom gcode ones), the config is validated/saved/dunno and the UI get refreshed

It's not so obvious if I do a copy/paste, but it became really annoying when I type something (say, a comment), the shortcuts UI are refreshed for evrey key press and locked until finished

* _Steps needed to reproduce the problem_
- Fire Slic3r
- Go to a setting tab and type something in a text area (custom Gcode or post processing scripts)

* _Expected Results_
Slic3r should refresh the UI only when the text area loose the focus (and only if something was changed inside)

* _Actual Results_
Slic3r refresh the UI at every KeyDown or KeyUp event

  * _Screenshots from __*Slic3r*__ preview are preferred_
Hard to take :)

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_
not relevant

gege2b, 2018-04-17:
I don't get why is not sufficient ? sorry if this is a dumb question, I suggested to use a "focusout" event because it seems logic to me

It's certainly not perfect, but would at least avoid a UI lockup for every chars typed, and verif/validation would occurs only when we leave that area, witch could mean that we have done with these settings

Also, what is the point to validate all settings while typing something in these text area ?
maybe these settings can just be validated/saved when closing the settings windows, given that post processing scripts or custom GCode aren't directly involved in flow calculation, validation is not critical here, am I wrong ?

RafaelEstevamReis, 2018-04-17:
In our software, we memorize the value in GotFocus and then compare in
LostFocus to trigger only when the value change

lordofhyphens, 2018-04-17:
I will try it out and see how it works.

The reason it verified everything before was an architectural quirk. I don't think it has to do that for any technical reasons.

lordofhyphens, 2018-04-18:
@gege2b I've pushed a commit that should reduce the cost of _update in Print settings (only!). Please give it a try and see if it helps. If it does, I'll look into doing more of it.

foreachthing, 2018-04-18:
@lordofhyphens Slic3r still crashes when removing the post processing script path.
The screen flickers a lot when adding notes or change the post processing script.

gege2b, 2018-04-18:
OK, two interesting things. I'm testing on another computer with Linux Mint Serena (KDE)

- BEFORE the git pull (commit c3cb01e ), I don't have the problem at all... the shortcuts are updated with no glitches, very smoothly
- AFTER the git pull (so, c206ae7 ) there is no visible difference, still smooth and glitche-free, so at least the fix didn't include a regression :-)

I will try tomorrow on my Win10 at work, where the problem was clearly visible

and @foreachthing is right, Slic3r crash if I change the post processing path in the shortcut (same on Win10), but I will open another issue  for this one, because it's a different bug

lordofhyphens, 2018-04-18:
I think this is also related to #4010

gege2b, 2018-04-19:
Just tested on my Win10 PC and still get the same behaviour

I recorded a screencap using VLC so you can see what it does (saved as zip so I cna upload it to Github)

[Slic3r.zip](https://github.com/slic3r/Slic3r/files/1927170/Slic3r.zip)

Oh, and it doest this with every setting, but it's far more annoying with PPS or Custom GCode

Bluelinegecko, 2018-10-12:
Spent some time trying to trouble shoot this tonight. I've narrowed down my issue to the configuration bundle (.ini) in Windows10.

I kept trying newer Dev versions hoping the bug would be fixed and kept experiencing the same problem. I noticed that each time, although I was running from a separate directory, slic3r still had my old configuration settings saved each time I opened it up. I thought it might be related to the slic3r bundled with repetier-host so I uninstalled that as well. Eventually I found out the settings were stored in the App data folder as mentioned in this post

http://www.soliforum.com/topic/3263/how-to-completely-uninstall-slic3r/

After deleting the files and testing the 09/08/18-64-bit dev version of slic3r everything was extremely responsive again and functioned great. As soon as I imported the configuration bundle I had backed up before deleting everything, the extreme sluggishness and constant refreshing of the UI occurred again.

I have a LOT of custom print/filament settings and hate to lose them all and recreate each one from scratch. I suspect that the addition of the new data fields that did not have settings stored when they were created in earlier versions of slic3r might have something to do with this. Any thought? Coding is not my strong point.

lordofhyphens, 2018-10-12:
Anything that is not present in a config is given the default.

If there is any sluggishness it is probably from the number of config
files. Probably would have to do some profiling on a system with sluggish
UI and figure out where the hot spots are.

lordofhyphens, 2018-10-12:
As a FYI, the files in the app data folder can be moved on a per-file basis (each is one profile)

Bluelinegecko, 2018-10-12:
That's good to know. I'll try archiving some of my less used profiles to cut down on the bulk and see how things perform. I still notice a fast UI refresh when starting with a fresh build and no imported config files, but it doesn't hang for 5-10 seconds for every key press like it did before. I also haven't seen it mysteriously trying to reslice the file automatically in the background either which is what was really frustrating.

Bluelinegecko, 2018-10-12:
Question: If they are all separate files, "should" the number of config files present on a user's system cause a slow down? I would assume that the program would only be concerned with what 3 profiles are currently selected and not need to be accessing the rest of the data except momentarily when changing from one profile to the other.

Bluelinegecko, 2018-10-24:
So after cleaning out some of my unused profiles I definitely see a good bit of a reduction in the lag I was experiencing. It's still there, but not as bad. If there is some way to reduce the effect of having a lot of profiles loaded it would be great. I have a lot of partially used filament spools of different types/colors and print settings between my two printers.

More importantly, I think I discovered the cause of the extreme lag and "automatic" slicing I referenced earlier. I noticed that it only seems to be happening when I have the "preview" tab selected for the platter instead of the 3D view. 

If I switch to the "preview tab" slic3r will go ahead and slice the model. This might be "by design" but I don't remember it working this way before. I thought I remember having to hit the "export g-code" button before it would slice the file and if I switched to the preview tab without  hitting the "export g-code" button first it would just show a blank screen. I could be recalling incorrectly.

Usually, If I am making a change to one or more settings it is after looking at the layer preview. I might adjust the infill, support settings, extrusion width, etc. Now it hangs on each key-press when changing certain printer/filament/print settings as it appears it is trying to re-slice the file while the preview tab is still open. If I switch the platter view to the 3D, 2D, or layers tab first, before making the changes to the profiles, there is still a noticeable UI refresh, but not as much of a delay. When changing some profile settings like perimeters/infill, when I return to the platter window the program will have automatically returned to the default 3d view. However, other times, upon changing some settings like the "speed" settings in the print profile tab, there is a longer hang up and when I return to the platter view it is sill showing the preview tab.

 I keep "Background processing" switched OFF in the preferences so I expect it should only slice the file when I hit "Export" or maybe even when I switch to the "Preview" tab, but not keep trying to re-slice after any input/changes while the tab is still open.

Hope I explained it clearly enough. It would be interesting if anyone else who commented on this thread is experiencing the same thing, and or if they noticed a reduction in the lag when reducing the number of profiles. I hope I am not inadvertently hijacking this thread.

Build commit: d17659b67c65bb5003ec92bf0d56ed7f8810e970