slic3r/Slic3r #4386: Changing a post processing path in a shortcut crash slic3r

gege2b, 2018-04-18:
### Version
c206ae77a3c967ec9d5d4595975149d588083b71

### Operating system type + version
Linux Mint Serena (KDE) but the crash also occurs on Win10 with a (not so) older commit

### Behavior
* _Describe the problem_
when testing for the #4385, I found this other bug.
When the post processing path area is shown on the shortcuts, changing something inside will just crash Slic3r 

* _Steps needed to reproduce the problem_
- Fire Slic3r
- Open print settings, type something into post processing script area (whatever you want) so the text area appears on the shortcuts
- close print settings
- change something in the post process. area
- kaboom!

* _Expected Results_
No crash

* _Actual Results_
Crash

  * _Screenshots from __*Slic3r*__ preview are preferred_
No screenshot but the console output : 

```
Error in function ConfigBase__set at src/perlglue.cpp:182: Trying to set non-existing option
         at /home/gege/Documents/build/Slic3r/lib/Slic3r/GUI/Plater.pm line 406.
        Slic3r::GUI::Plater::__ANON__("post_process#0") called at /home/gege/Documents/build/Slic3r/lib/Slic3r/GUI/Plater/OverrideSettingsPanel.pm line 168
        Slic3r::GUI::Plater::OverrideSettingsPanel::__ANON__("post_process#0", "dfdsfsdfdsfd") called at /home/gege/Documents/build/Slic3r/lib/Slic3r/GUI/OptionsGroup.pm line 258
        Slic3r::GUI::OptionsGroup::_on_change(Slic3r::GUI::ConfigOptionsGroup=HASH(0x5557ea0), "post_process#0", "dfdsfsdfdsfd") called at /home/gege/Documents/build/Slic3r/lib/Slic3r/GUI/OptionsGroup.pm line 475
        Slic3r::GUI::ConfigOptionsGroup::_on_change(Slic3r::GUI::ConfigOptionsGroup=HASH(0x5557ea0), "post_process#0", "dfdsfsdfdsfd") called at /home/gege/Documents/build/Slic3r/lib/Slic3r/GUI/OptionsGroup.pm line 162
        Slic3r::GUI::OptionsGroup::__ANON__("post_process#0", "dfdsfsdfdsfd") called at /home/gege/Documents/build/Slic3r/lib/Slic3r/GUI/OptionsGroup/Field.pm line 36
        Slic3r::GUI::OptionsGroup::Field::_on_change(Slic3r::GUI::OptionsGroup::Field::TextCtrl=HASH(0x5568a88), "post_process#0") called at /home/gege/Documents/build/Slic3r/lib/Slic3r/GUI/OptionsGroup/Field.pm line 192
        Slic3r::GUI::OptionsGroup::Field::TextCtrl::__ANON__(Slic3r::GUI::Plater::OverrideSettingsPanel=HASH(0x45fb5e8), Wx::CommandEvent=SCALAR(0x554d1f8)) called at ./slic3r.pl line 126
        eval {...} called at ./slic3r.pl line 126
```

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_
not necessary

foreachthing, 2018-04-19:
similar to https://github.com/slic3r/Slic3r/issues/4323. _(just for ref.)_

gege2b, 2018-04-19:
Seems similar indeed, sorry for the dupe
I'll close this and post back on your original issue

closed 2018-04-19 by gege2b