slic3r/Slic3r #4389: [Feature Request] Support XY-Offset

foreachthing, 2018-04-23:
### Version
Any

### Operating system type + version
Windows 7 64bit

### Behavior
* When using Support inside a hole, the support is way too close to the geometry (all widths set to default; ø 0.4 mm nozzle). I'm not able to remove the support from inside this hole (part attached).
* Also, this hole has a small shoulder and the support _grows_ behind this shoulder which makes it even harder to remove it.

### myWish
* I wish for a **Print Setting**, where I can set the XY-offset (mm, line width, % ...) of the support from the geometry.
* Support should not grow _inside geometry_.
* In this case: support should be trimmed to first layer projection for inside-geometry.

_Is this a new feature request?_
Yes.
_In Cura there are settings available to adjust XY-offset. I had to print this part with Cura to get removable support :-(_

#### STL/Config (.ZIP) where problem occurs
* [part.zip](https://github.com/slic3r/Slic3r/files/1937304/part.zip)
* Config: use support.

lordofhyphens, 2018-04-23:
This should already be a thing, although it may be a constant that can be
tweaked in SupportMaterial.pm.

foreachthing, 2018-04-23:
I think I got it.
Yes, it's constant.
It's in said file in `sub clip_with_object {}`. I changed this line ` offset([ map @$_, map @{$_->slices}, @layers ], +$self->flow->scaled_width),` to ` offset([ map @$_, map @{$_->slices}, @layers ], + 3 * $self->flow->scaled_width),` and it's much, much better now (for this case!).
Still ... any chance to make that a print setting?

lordofhyphens, 2018-04-23:
Yeah, should be straightforward.  I think I did not because I was less
interested in spending any time on perl support code and people were very
likely to get into trouble with that setting.

lordofhyphens, 2018-05-05:
@foreachthing https://github.com/slic3r/Slic3r/wiki/Code%3A-Adding-new-option-to-GUI-CLI if you want to take a swing at it yourself to try it out and see what breaks.