slic3r/Slic3r #4394: Missing environment variables ie. SLIC3R_LAYER_HEIGHT - python script

HashDefineElectronics, 2018-05-03:
### Version
1.2.9

### Operating system type + version
Ubuntu 16.04.4 LTS

### Behavior
Not all Slic3r configuration options are passed as environmental variables to the post script. 

This is what the script  extracted from the environmental variables.
```
SLIC3R_OUTPUT_FILENAME_FORMAT [input_filename_base].gcode
SLIC3R_FIRST_LAYER_SPEED 30
SLIC3R_FIRST_LAYER_ACCELERATION 0
SLIC3R_RETRACT_BEFORE_TRAVEL 0,2
SLIC3R_RETRACT_LENGTH 0.5,0.5
SLIC3R_USE_FIRMWARE_RETRACTION 0
SLIC3R_PERIMETER_ACCELERATION 0
SLIC3R_RETRACT_RESTART_EXTRA_TOOLCHANGE 0,0
SLIC3R_ONLY_RETRACT_WHEN_CROSSING_PERIMETERS 1
SLIC3R_SKIRT_HEIGHT 1
SLIC3R_SKIRT_DISTANCE 6
SLIC3R_EXTRUSION_AXIS E
SLIC3R_MIN_FAN_SPEED 100
SLIC3R_RETRACT_SPEED 30,30
SLIC3R_FIRST_LAYER_EXTRUSION_WIDTH 200%
SLIC3R_FIRST_LAYER_TEMPERATURE 210,210
SLIC3R_WIPE 0,0
SLIC3R_USE_VOLUMETRIC_E 0
SLIC3R_USE_RELATIVE_E_DISTANCES 1
SLIC3R_START_GCODE 
SLIC3R_END_GCODE 
SLIC3R_BRIDGE_ACCELERATION 0
SLIC3R_TEMPERATURE 195,195
SLIC3R_DEFAULT_ACCELERATION 0
SLIC3R_EXTRUDER_CLEARANCE_HEIGHT 20
SLIC3R_BRIM_WIDTH 0
SLIC3R_MIN_SKIRT_LENGTH 0
SLIC3R_BEFORE_LAYER_GCODE 
SLIC3R_LAYER_GCODE 
SLIC3R_NOTES 
SLIC3R_SLOWDOWN_BELOW_LAYER_TIME 5
SLIC3R_FILAMENT_COLOUR #3159FF;#3159FF
SLIC3R_Z_OFFSET 0
SLIC3R_NOZZLE_DIAMETER 0.3,0.8
SLIC3R_GCODE_COMMENTS 0
SLIC3R_RETRACT_LAYER_CHANGE 0,0
SLIC3R_MAX_FAN_SPEED 100
SLIC3R_STANDBY_TEMPERATURE_DELTA -5
SLIC3R_VIBRATION_LIMIT 0
SLIC3R_FAN_ALWAYS_ON 1
SLIC3R_TRAVEL_SPEED 200
SLIC3R_FILAMENT_DIAMETER 1.75,1.75
SLIC3R_GCODE_ARCS 0
SLIC3R_FIRST_LAYER_BED_TEMPERATURE 60
SLIC3R_BED_TEMPERATURE 50
SLIC3R_COOLING 1
SLIC3R_MAX_PRINT_SPEED 30
SLIC3R_COMPLETE_OBJECTS 0
SLIC3R_BED_SHAPE 0x0,200x0,200x200,0x200
SLIC3R_EXTRUDER_CLEARANCE_RADIUS 20
SLIC3R_DUPLICATE_DISTANCE 6
SLIC3R_SPIRAL_VASE 0
SLIC3R_RESOLUTION 0
SLIC3R_RETRACT_LIFT 0,0
SLIC3R_PRESSURE_ADVANCE 0
SLIC3R_AVOID_CROSSING_PERIMETERS 0
SLIC3R_EXTRUDER_OFFSET 0x0,0x0
SLIC3R_BRIDGE_FAN_SPEED 100
SLIC3R_DISABLE_FAN_FIRST_LAYERS 3
SLIC3R_GCODE_FLAVOR reprap
SLIC3R_INFILL_ACCELERATION 0
SLIC3R_THREADS 2
SLIC3R_RETRACT_RESTART_EXTRA 0,0
SLIC3R_FAN_BELOW_LAYER_TIME 60
SLIC3R_RETRACT_LENGTH_TOOLCHANGE 10,10
SLIC3R_TOOLCHANGE_GCODE 
SLIC3R_SKIRTS 1
SLIC3R_MIN_PRINT_SPEED 10
SLIC3R_POST_PROCESS /home/user/Documents/Slicer_post_script/start.py
SLIC3R_INFILL_FIRST 0
SLIC3R_EXTRUSION_MULTIPLIER 1,1
SLIC3R_OOZE_PREVENTION 0
SLIC3R_MAX_VOLUMETRIC_SPEED 0

```

### Additional notes
- The script was writing in Python (2.7).

#### STL/Config (.ZIP) where problem occurs
[config.ini.zip](https://github.com/slic3r/Slic3r/files/1970429/config.ini.zip)

lordofhyphens, 2018-05-03:
I am not sure it is as useful as you think, given that it can change from one part of the model to another.

lordofhyphens, 2018-05-04:
Layer height is not something that is necessarily static for a print; it can be overridden in multiple ways (adaptive slicing, there's z height table, etc). Thus, such a variable is not suitable for an entire print.

I could probably enhance the verbose gcode output to put the expected layer height in a comment instead.

lordofhyphens, 2018-05-04:
[4394.zip](https://github.com/slic3r/Slic3r/files/1975750/4394.zip) is a dump of everything needed to check a config export against the environment print. I have not seen what looks like a pattern to me (yet). 

https://github.com/slic3r/Slic3r/blob/master/xs/src/libslic3r/Config.cpp#L362 is the area of code that actually sets the environment variables. There's not much here that can go wrong :/ 

This function calls through to keys() which builds a list of all of the option names that should be stored in the config.

HashDefineElectronics, 2018-05-08:
Thank you @lordofhyphens 

> Layer height is not something that is necessarily static for a print; it can be overridden in multiple ways (adaptive slicing, there's z height table, etc). Thus, such a variable is not suitable for an entire print.

I see that makes sense. I should be able to workout the layer height(s) from the gcode. to be honest I'm just concern that if I was missing layer height from the env variables then is there an issue I should be worried about?

I gave your config and post script ago and i've gotten a few messages slicer about ***Unknown option***. I wonder if this is anything to do with my issue. Also, the config and script didn't return the layer height either. :S

```
./Slic3r/bin/slic3r --load /home/Bob/Documents/post_script/4394/config.ini /home/Bob/Documents/post_script/4394/20mmbox.stl
Unknown option match_horizontal_surfaces
Unknown option retract_lift_below
Unknown option shortcuts
Unknown option use_set_and_wait_extruder
Unknown option sequential_print_priority
Unknown option support_material_interface_extrusion_width
Unknown option end_filament_gcode
Unknown option print_host
Unknown option interior_brim_width
Unknown option filament_density
Unknown option brim_connections_width
Unknown option compatible_printers
Unknown option min_layer_height
Unknown option between_objects_gcode
Unknown option printer_notes
Unknown option top_infill_pattern
Unknown option z_steps_per_mm
Unknown option start_filament_gcode
Unknown option adaptive_slicing
Unknown option regions_overlap
Unknown option serial_port
Unknown option adaptive_slicing_quality
Unknown option print_settings_id
Unknown option use_set_and_wait_bed
Unknown option filament_notes
Unknown option serial_speed
Unknown option support_material_buildplate_only
Unknown option filament_max_volumetric_speed
Unknown option host_type
Unknown option filament_cost
Unknown option filament_settings_id
Unknown option max_layer_height
Unknown option has_heatbed
Unknown option support_material_max_layers
Unknown option retract_lift_above
Unknown option bottom_infill_pattern
Unknown option printer_settings_id
Unknown option fill_gaps
=> Processing triangulated mesh
=> Generating perimeters
=> Preparing infill
=> Infilling layers
=> Generating skirt
=> Exporting G-code to /home/bob/Documents/post_script/4394/20mmbox.gcode
=> Running post-processing scripts
Done. Process took 0 minutes and 1.467 seconds
Filament required: 745.5mm (1.7cm3)

```

I'm currently on 1.2.9 so I'm going to give 1.3 ago and see what happens.

lordofhyphens, 2018-05-08:
Yeah there are certainly options that look missing in 1.3.0-dev.

You are seeing a bunch of missing option messages in 1.2.9 because those
options didn't exist in 1.2.9.

HashDefineElectronics, 2018-05-08:
Just given 1.3.0-dev ago and it seem to do the job.

It still doesn't put out layer_height but base on what you said, it make perfect sense that it doesn't. It does have min & max layer height which is perfect.

Thank you @lordofhyphens for the help.