slic3r/Slic3r #4395: Slic3r Build fails after perl Build.PL command

jclinton830, 2018-05-04:
### Version
_1.29_

### Operating system type + version
_Win 7 32 bit_

I am completely new to the development of Slic3r, I hope someone can help me set this up. I have been struggling for a few days and have managed to fix a lot of errors but not this particular one.

### Behavior
* _ After running perl Build.PL command it fails with an error building xs.dll... here is the complete output after execution perl Build.PL command_

`App::cpanminus is up to date. (1.7044)
Class::XSAccessor is up to date. (1.19)
Encode::Locale is up to date. (1.05)
ExtUtils::MakeMaker is up to date. (7.34)
ExtUtils::ParseXS is up to date. (3.35)
File::Basename is up to date. (2.85)
File::Spec is up to date. (3.74)
Getopt::Long is up to date. (2.5)
IO::Scalar is up to date. (2.111)
Math::PlanePath is up to date. (126)
Module::Build::WithXSpp is up to date. (0.14)
Moo is up to date. (2.003004)
Scalar::Util is up to date. (1.50)
Test::Harness is up to date. (3.42)
Test::More is up to date. (1.302136)
Time::HiRes is up to date. (1.9758)
Could not find the MinGW directory at C:\dev\CitrusPerl\mingw32; skipping XML::SAX::ExpatXS (only needed for faster parsing of AMF files)
XML::SAX::ExpatXS is up to date. (1.33)
Cleaning up build files
Cleaning up configuration files
--> Working on ./xs
Configuring C:/Development/DalekSlic3r/Slic3r/xs ... Created MYMETA.yml and MYMETA.json
Creating new 'Build' script for 'Slic3r-XS' version '0.01'
OK
Building and testing Slic3r-XS-0.01 ... Building Slic3r-XS
Processing XS typemap files...
Generating main XS file...
gcc -c -s -O2 -DWIN32 -D__USE_MINGW_ANSI_STDIO -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DUSE_PERLIO -fwrapv -fno-strict-aliasing -mms-bitfields -s -O2 -DWIN32 -D__USE_MINGW_ANSI_STDIO -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DUSE_PERLIO -fwrapv -fno-strict-aliasing -mms-bitfields -xc++ -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -Isrc -Ibuildtmp -s -O2 -I"C:\STRAWB~1\perl\lib\CORE" -I"C:\STRAWB~1\c\include" -o "src\admesh\connect.o" "src\admesh\connect.c"
gcc -c -s -O2 -DWIN32 -D__USE_MINGW_ANSI_STDIO -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DUSE_PERLIO -fwrapv -fno-strict-aliasing -mms-bitfields -s -O2 -DWIN32 -D__USE_MINGW_ANSI_STDIO -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DUSE_PERLIO -fwrapv -fno-strict-aliasing -mms-bitfields -xc++ -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -Isrc -Ibuildtmp -s -O2 -I"C:\STRAWB~1\perl\lib\CORE" -I"C:\STRAWB~1\c\include" -o "src\Extruder.o" "src\Extruder.cpp"
gcc -c -s -O2 -DWIN32 -D__USE_MINGW_ANSI_STDIO -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DUSE_PERLIO -fwrapv -fno-strict-aliasing -mms-bitfields -s -O2 -DWIN32 -D__USE_MINGW_ANSI_STDIO -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DUSE_PERLIO -fwrapv -fno-strict-aliasing -mms-bitfields -xc++ -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -Isrc -Ibuildtmp -s -O2 -I"C:\STRAWB~1\perl\lib\CORE" -I"C:\STRAWB~1\c\include" -o "src\Line.o" "src\Line.cpp"
gcc -c -s -O2 -DWIN32 -D__USE_MINGW_ANSI_STDIO -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DUSE_PERLIO -fwrapv -fno-strict-aliasing -mms-bitfields -s -O2 -DWIN32 -D__USE_MINGW_ANSI_STDIO -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DUSE_PERLIO -fwrapv -fno-strict-aliasing -mms-bitfields -xc++ -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -Isrc -Ibuildtmp -s -O2 -I"C:\STRAWB~1\perl\lib\CORE" -I"C:\STRAWB~1\c\include" -o "src\PrintConfig.o" "src\PrintConfig.cpp"
gcc -c -s -O2 -DWIN32 -D__USE_MINGW_ANSI_STDIO -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DUSE_PERLIO -fwrapv -fno-strict-aliasing -mms-bitfields -s -O2 -DWIN32 -D__USE_MINGW_ANSI_STDIO -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DUSE_PERLIO -fwrapv -fno-strict-aliasing -mms-bitfields -xc++ -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -Isrc -Ibuildtmp -s -O2 -I"C:\STRAWB~1\perl\lib\CORE" -I"C:\STRAWB~1\c\include" -o "src\Surface.o" "src\Surface.cpp"
gcc -c -s -O2 -DWIN32 -D__USE_MINGW_ANSI_STDIO -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DUSE_PERLIO -fwrapv -fno-strict-aliasing -mms-bitfields -s -O2 -DWIN32 -D__USE_MINGW_ANSI_STDIO -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DUSE_PERLIO -fwrapv -fno-strict-aliasing -mms-bitfields -xc++ -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -Isrc -Ibuildtmp -s -O2 -I"C:\STRAWB~1\perl\lib\CORE" -I"C:\STRAWB~1\c\include" -o "src\Model.o" "src\Model.cpp"
gcc -c -s -O2 -DWIN32 -D__USE_MINGW_ANSI_STDIO -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DUSE_PERLIO -fwrapv -fno-strict-aliasing -mms-bitfields -s -O2 -DWIN32 -D__USE_MINGW_ANSI_STDIO -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DUSE_PERLIO -fwrapv -fno-strict-aliasing -mms-bitfields -xc++ -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -Isrc -Ibuildtmp -s -O2 -I"C:\STRAWB~1\perl\lib\CORE" -I"C:\STRAWB~1\c\include" -o "src\utils.o" "src\utils.cpp"
gcc -c -s -O2 -DWIN32 -D__USE_MINGW_ANSI_STDIO -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DUSE_PERLIO -fwrapv -fno-strict-aliasing -mms-bitfields -s -O2 -DWIN32 -D__USE_MINGW_ANSI_STDIO -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DUSE_PERLIO -fwrapv -fno-strict-aliasing -mms-bitfields -xc++ -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -Isrc -Ibuildtmp -s -O2 -I"C:\STRAWB~1\perl\lib\CORE" -I"C:\STRAWB~1\c\include" -o "src\admesh\util.o" "src\admesh\util.c"
gcc -c -s -O2 -DWIN32 -D__USE_MINGW_ANSI_STDIO -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DUSE_PERLIO -fwrapv -fno-strict-aliasing -mms-bitfields -s -O2 -DWIN32 -D__USE_MINGW_ANSI_STDIO -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DUSE_PERLIO -fwrapv -fno-strict-aliasing -mms-bitfields -xc++ -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -Isrc -Ibuildtmp -s -O2 -I"C:\STRAWB~1\perl\lib\CORE" -I"C:\STRAWB~1\c\include" -o "src\ExtrusionEntityCollection.o" "src\ExtrusionEntityCollection.cpp"
gcc -c -s -O2 -DWIN32 -D__USE_MINGW_ANSI_STDIO -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DUSE_PERLIO -fwrapv -fno-strict-aliasing -mms-bitfields -s -O2 -DWIN32 -D__USE_MINGW_ANSI_STDIO -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DUSE_PERLIO -fwrapv -fno-strict-aliasing -mms-bitfields -xc++ -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -Isrc -Ibuildtmp -s -O2 -I"C:\STRAWB~1\perl\lib\CORE" -I"C:\STRAWB~1\c\include" -o "src\poly2tri\sweep\sweep_context.o" "src\poly2tri\sweep\sweep_context.cc"
gcc -c -s -O2 -DWIN32 -D__USE_MINGW_ANSI_STDIO -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DUSE_PERLIO -fwrapv -fno-strict-aliasing -mms-bitfields -s -O2 -DWIN32 -D__USE_MINGW_ANSI_STDIO -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DUSE_PERLIO -fwrapv -fno-strict-aliasing -mms-bitfields -xc++ -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -Isrc -Ibuildtmp -s -O2 -I"C:\STRAWB~1\perl\lib\CORE" -I"C:\STRAWB~1\c\include" -o "src\ExPolygonCollection.o" "src\ExPolygonCollection.cpp"
gcc -c -s -O2 -DWIN32 -D__USE_MINGW_ANSI_STDIO -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DUSE_PERLIO -fwrapv -fno-strict-aliasing -mms-bitfields -s -O2 -DWIN32 -D__USE_MINGW_ANSI_STDIO -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DUSE_PERLIO -fwrapv -fno-strict-aliasing -mms-bitfields -xc++ -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -Isrc -Ibuildtmp -s -O2 -I"C:\STRAWB~1\perl\lib\CORE" -I"C:\STRAWB~1\c\include" -o "src\Config.o" "src\Config.cpp"
gcc -c -s -O2 -DWIN32 -D__USE_MINGW_ANSI_STDIO -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DUSE_PERLIO -fwrapv -fno-strict-aliasing -mms-bitfields -s -O2 -DWIN32 -D__USE_MINGW_ANSI_STDIO -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DUSE_PERLIO -fwrapv -fno-strict-aliasing -mms-bitfields -xc++ -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -Isrc -Ibuildtmp -s -O2 -I"C:\STRAWB~1\perl\lib\CORE" -I"C:\STRAWB~1\c\include" -o "src\poly2tri\sweep\advancing_front.o" "src\poly2tri\sweep\advancing_front.cc"
[... cut ...]
Checking if you have Test::More 0 ... Yes (1.302136)
Checking if you have ExtUtils::ParseXS 3.18 ... Yes (3.35)
Checking if you have ExtUtils::Typemap 1.00 ... Yes (1.00)
Checking if you have Module::Build 0.3601 ... Yes (0.4224)
Checking if you have ExtUtils::XSpp 0.18 ... Yes (0.18)
Building and testing Slic3r-XS-0.01
-> FAIL Installing ./xs failed. See C:\Users\ipri\.cpanm\work\1525393610.1604\build.log for details. Retry with --force to force install it.
`

I am running Stawberry-perl 32 bit

lordofhyphens, 2018-05-04:
Replace your Strawberry directory with the contents of
https://bintray.com/lordofhyphens/Slic3r/download_file?file_path=Strawberry-6.3.0-seg-archive.7z
Extract 
https://bintray.com/lordofhyphens/Slic3r/download_file?file_path=boost_1_63_0-x32-gcc-4.9.2-sjlj.7z

to C:\dev.
Extract https://bintray.com/lordofhyphens/Slic3r/download_file?file_path=local-lib-32bit.7z to the local-lib dir in your slic3r checkout path. 

https://bintray.com/lordofhyphens/Slic3r/download_file?file_path=freeglut-mingw-3.0.0.32bit.7z should go in the FreeGLUT directory in your Slic3r checkout. 

Never run perl Build.pl --gui.

jclinton830, 2018-05-04:
I believe the stawberry you provided is a 64 bit version.

lordofhyphens, 2018-05-04:
Sorry about that.

http://strawberryperl.com/download/5.24.1.1/strawberry-perl-5.24.1.1-32bit.msi

lordofhyphens, 2018-05-04:
Install local::lib and the contents of local-lib dir and you should sidestep most perl related install garbage.

jclinton830, 2018-05-07:
@lordofhyphens Tried with the 32 bit version of strawberry perl and did not work. I have tried building Slic3r on a 64 bit win 10 and I am stuck in the same place too. I have tried everything you said in the same order but still no luck. Do you believe there are problems with the source code's last commit?

So far I have only been able to build on an ubuntu machine. However, I would like to set this up in windows. 

Thanks.

MyPrinter7, 2018-05-07:
Hi both, 

I believe I'm having the same issue here... I'm trying to build on 64-bit windows 10, and the build fails after Perl Build.PL command with the following error...

`App::cpanminus is up to date. (1.7044)
Class::XSAccessor is up to date. (1.19)
Encode::Locale is up to date. (1.05)
ExtUtils::MakeMaker is up to date. (7.34)
ExtUtils::ParseXS is up to date. (3.35)
File::Basename is up to date. (2.85)
File::Spec is up to date. (3.74)
Getopt::Long is up to date. (2.5)
IO::Scalar is up to date. (2.111)
Math::PlanePath is up to date. (126)
Module::Build::WithXSpp is up to date. (0.14)
Moo is up to date. (2.003004)
Scalar::Util is up to date. (1.50)
Test::Harness is up to date. (3.42)
Test::More is up to date. (1.302136)
Time::HiRes is up to date. (1.9758)
Could not find the MinGW directory at C:\dev\CitrusPerl\mingw32; skipping XML::SAX::ExpatXS (only needed for faster parsing of AMF files)
XML::SAX::ExpatXS is up to date. (1.33)
Cleaning up build files
Cleaning up configuration files
--> Working on ./xs
Configuring C:/Development/Slic3r/Slic3r/xs ... Created MYMETA.yml and MYMETA.json
Creating new 'Build' script for 'Slic3r-XS' version '0.01'
OK
Building and testing Slic3r-XS-0.01 ... Building Slic3r-XS
Processing XS typemap files...
Generating main XS file...
gcc -c -s -O2 -DWIN32 -DWIN64 -DCONSERVATIVE -D__USE_MINGW_ANSI_STDIO -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DUSE_PERLIO -fwrapv -fno-strict-aliasing -mms-bitfields -s -O2 -DWIN32 -DWIN64 -DCONSERVATIVE -D__USE_MINGW_ANSI_STDIO -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DUSE_PERLIO -fwrapv -fno-strict-aliasing -mms-bitfields -xc++ -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -Isrc -Ibuildtmp -s -O2 -I"C:\STRAWB~1\perl\lib\CORE" -I"C:\STRAWB~1\c\include" -o "src\admesh\shared.o" "src\admesh\shared.c"
gcc -c -s -O2 -DWIN32 -DWIN64 -DCONSERVATIVE -D__USE_MINGW_ANSI_STDIO -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DUSE_PERLIO -fwrapv -fno-strict-aliasing -mms-bitfields -s -O2 -DWIN32 -DWIN64 -DCONSERVATIVE -D__USE_MINGW_ANSI_STDIO -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DUSE_PERLIO -fwrapv -fno-strict-aliasing -mms-bitfields -xc++ -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -Isrc -Ibuildtmp -s -O2 -I"C:\STRAWB~1\perl\lib\CORE" -I"C:\STRAWB~1\c\include" -o "src\SVG.o" "src\SVG.cpp"
gcc -c -s -O2 -DWIN32 -DWIN64 -DCONSERVATIVE -D__USE_MINGW_ANSI_STDIO -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DUSE_PERLIO -fwrapv -fno-strict-aliasing -mms-bitfields -s -O2 -DWIN32 -DWIN64 -DCONSERVATIVE -D__USE_MINGW_ANSI_STDIO -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DUSE_PERLIO -fwrapv -fno-strict-aliasing -mms-bitfields -xc++ -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -Isrc -Ibuildtmp -s -O2 -I"C:\STRAWB~1\perl\lib\CORE" -I"C:\STRAWB~1\c\include" -o "src\admesh\stlinit.o" "src\admesh\stlinit.c"
gcc -c -s -O2 -DWIN32 -DWIN64 -DCONSERVATIVE -D__USE_MINGW_ANSI_STDIO -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DUSE_PERLIO -fwrapv -fno-strict-aliasing -mms-bitfields -s -O2 -DWIN32 -DWIN64 -DCONSERVATIVE -D__USE_MINGW_ANSI_STDIO -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DUSE_PERLIO -fwrapv -fno-strict-aliasing -mms-bitfields -xc++ -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -Isrc -Ibuildtmp -s -O2 -I"C:\STRAWB~1\perl\lib\CORE" -I"C:\STRAWB~1\c\include" -o "src\SurfaceCollection.o" "src\SurfaceCollection.cpp"
gcc -c -s -O2 -DWIN32 -DWIN64 -DCONSERVATIVE -D__USE_MINGW_ANSI_STDIO -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DUSE_PERLIO -fwrapv -fno-strict-aliasing -mms-bitfields -s -O2 -DWIN32 -DWIN64 -DCONSERVATIVE -D__USE_MINGW_ANSI_STDIO -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DUSE_PERLIO -fwrapv -fno-strict-aliasing -mms-bitfields -xc++ -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -Isrc -Ibuildtmp -s -O2 -I"C:\STRAWB~1\perl\lib\CORE" -I"C:\STRAWB~1\c\include" -o "src\Extruder.o" "src\Extruder.cpp"
gcc -c -s -O2 -DWIN32 -DWIN64 -DCONSERVATIVE -D__USE_MINGW_ANSI_STDIO -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DUSE_PERLIO -fwrapv -fno-strict-aliasing -mms-bitfields -s -O2 -DWIN32 -DWIN64 -DCONSERVATIVE -D__USE_MINGW_ANSI_STDIO -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DUSE_PERLIO -fwrapv -fno-strict-aliasing -mms-bitfields -xc++ -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -Isrc -Ibuildtmp -s -O2 -I"C:\STRAWB~1\perl\lib\CORE" -I"C:\STRAWB~1\c\include" -o "src\poly2tri\sweep\sweep_context.o" "src\poly2tri\sweep\sweep_context.cc"
gcc -c -s -O2 -DWIN32 -DWIN64 -DCONSERVATIVE -D__USE_MINGW_ANSI_STDIO -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DUSE_PERLIO -fwrapv -fno-strict-aliasing -mms-bitfields -s -O2 -DWIN32 -DWIN64 -DCONSERVATIVE -D__USE_MINGW_ANSI_STDIO -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DUSE_PERLIO -fwrapv -fno-strict-aliasing -mms-bitfields -xc++ -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -Isrc -Ibuildtmp -s -O2 -I"C:\STRAWB~1\perl\lib\CORE" -I"C:\STRAWB~1\c\include" -o "src\poly2tri\common\shapes.o" "src\poly2tri\common\shapes.cc"
gcc -c -s -O2 -DWIN32 -DWIN64 -DCONSERVATIVE -D__USE_MINGW_ANSI_STDIO -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DUSE_PERLIO -fwrapv -fno-strict-aliasing -mms-bitfields -s -O2 -DWIN32 -DWIN64 -DCONSE [... cut ...]
Checking if you have ExtUtils::XSpp 0.18 ... Yes (0.18)
Checking if you have Test::More 0 ... Yes (1.302136)
Checking if you have Module::Build 0.3601 ... Yes (0.4224)
Building and testing Slic3r-XS-0.01
-> FAIL Installing ./xs failed. See C:\Users\user\.cpanm\work\1525669333.4980\build.log for details. Retry with --force to force install it.
`

I have 64-bit Strawberry Perl installed and Boost 1.63 installed successfully. @lordofhyphens Do you have any ideas why this might be happening? 

Thanks in advance.

lordofhyphens, 2018-05-07:
@jclinton830 
 https://ci.appveyor.com/project/lordofhyphens/slic3r says that it builds on Windows with the recommended distribution of Perl and the provided Boost. 

@MyPrinter7 you are going to have to skim through the build log and get the actual compiler error to figure out what is going on.

lordofhyphens, 2018-05-07:
@jclinton830 the exact copy of Boost used on the build server is available for download .

lordofhyphens, 2018-05-07:
Boost for 32 bit strawberry perl (gcc 4.9.2): https://bintray.com/lordofhyphens/Slic3r/download_file?file_path=boost_1_63_0-x32-gcc-4.9.2-sjlj.7z

Boost for custom 64-bit "slic3r perl" (gcc 6.3.0 with seh exceptions) http://www.siusgs.com/slic3r/buildserver/win/boost_1_63_0-x64-gcc-6.3.0-seh.7z

Your failures are in the build of Slic3r itself now, not any perl-specific dependencies. You should examine the full logs carefully for the errors from the compiler.

jclinton830, 2018-05-23:
Hi @lordofhyphens I gave up building Slic3r on that machine, been trying to build it in a windows 64 bit computer now. Somewhat similar outcome so far. I am running the Slic3r Perl provided in the github page, and boost 1.63 too. 

Boost built successfully with no problems, but Slic3r fails. Below is the CMD output:

`App::cpanminus is up to date. (1.7044)
Class::XSAccessor is up to date. (1.19)
Encode::Locale is up to date. (1.05)
ExtUtils::MakeMaker is up to date. (7.34)
ExtUtils::ParseXS is up to date. (3.35)
File::Basename is up to date. (2.85)
File::Spec is up to date. (3.74)
Getopt::Long is up to date. (2.5)
IO::Scalar is up to date. (2.111)
Math::PlanePath is up to date. (126)
Module::Build::WithXSpp is up to date. (0.14)
Moo is up to date. (2.003004)
Scalar::Util is up to date. (1.50)
Test::Harness is up to date. (3.42)
Test::More is up to date. (1.302136)
Time::HiRes is up to date. (1.9758)
Could not find the MinGW directory at C:\dev\CitrusPerl\mingw32; skipping XML::S
AX::ExpatXS (only needed for faster parsing of AMF files)
--> Working on XML::SAX::ExpatXS
Fetching http://www.cpan.org/authors/id/P/PC/PCIMPRICH/XML-SAX-ExpatXS-1.33.tar.
gz ... OK
Configuring XML-SAX-ExpatXS-1.33 ... OK
Building and testing XML-SAX-ExpatXS-1.33 ... FAIL
! Installing XML::SAX::ExpatXS failed. See C:\Users\ADMINI~1\.cpanm\work\1527054
680.2176\build.log for details. Retry with --force to force install it.
Don't worry, this module is optional.
Cleaning up build files
Cleaning up configuration files
cpanm (App::cpanminus) 1.7044 on perl 5.024001 built for MSWin32-x64-multi-threa
d
Work directory is C:\Users\ADMINI~1/.cpanm/work/1527054685.3396
You have make C:\Strawberry\c\bin\dmake.exe
You have LWP 6.24
Falling back to Archive::Tar 2.26
--> Working on ./xs
Entering C:/Development/Slic3r/xs
Checking if you have ExtUtils::Install 1.46 ... Yes (2.06)
Checking if you have Module::Build 0.38 ... Yes (0.4224)
Running Build.PL
Configuring C:/Development/Slic3r/xs ... 'cc' is not recognized as an internal o
r external command,
operable program or batch file.
Created MYMETA.yml and MYMETA.json
Creating new 'Build' script for 'Slic3r-XS' version '0.01'
OK
Checking dependencies from MYMETA.json ...
Checking if you have Module::Build 0.3601 ... Yes (0.4224)
Checking if you have ExtUtils::Typemap 1.00 ... Yes (1.00)
Checking if you have Test::More 0 ... Yes (1.302136)
Checking if you have ExtUtils::ParseXS 3.18 ... Yes (3.35)
Checking if you have ExtUtils::Typemaps::Default 1.03 ... Yes (1.05)
Checking if you have ExtUtils::XSpp 0.18 ... Yes (0.18)
Building and testing Slic3r-XS-0.01 ... Building Slic3r-XS
Processing XS typemap files...
Generating main XS file...
gcc -c -s -O2 -DWIN32 -DWIN64 -DCONSERVATIVE  -DPERL_TEXTMODE_SCRIPTS -DPERL_IMP
LICIT_CONTEXT -DPERL_IMPLICIT_SYS -fwrapv -fno-strict-aliasing -mms-bitfields -s
 -O2 -DWIN32 -DWIN64 -DCONSERVATIVE -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONT
EXT -DPERL_IMPLICIT_SYS -fwrapv -fno-strict-aliasing -mms-bitfields -xc++ -D_GLI
BCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -Isrc -Ibuildtmp -s -O2 -I"C:\STRAWB~
1\perl\lib\CORE" -I"C:\STRAWB~1\c\include" -o "src\Model.o" "src\Model.cpp"
gcc -c -s -O2 -DWIN32 -DWIN64 -DCONSERVATIVE  -DPERL_TEXTMODE_SCRIPTS -DPERL_IMP
LICIT_CONTEXT -DPERL_IMPLICIT_SYS -fwrapv -fno-strict-aliasing -mms-bitfields -s
 -O2 -DWIN32 -DWIN64 -DCONSERVATIVE -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONT
EXT -DPERL_IMPLICIT_SYS -fwrapv -fno-strict-aliasing -mms-bitfields -xc++ -D_GLI
BCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -Isrc -Ibuildtmp -s -O2 -I"C:\STRAWB~
[... 249 lines cut ...]
Checking if you have Module::Build 0.3601 ... Yes (0.4224)
Checking if you have ExtUtils::Typemap 1.00 ... Yes (1.00)
Checking if you have Test::More 0 ... Yes (1.302136)
Checking if you have ExtUtils::Typemaps::Default 1.03 ... Yes (1.05)
Checking if you have ExtUtils::XSpp 0.18 ... Yes (0.18)
Checking if you have ExtUtils::ParseXS 3.18 ... Yes (3.35)
Building and testing Slic3r-XS-0.01
-> FAIL Installing ./xs failed. See C:\Users\ADMINI~1\.cpanm\work\1527056148.4680\build.log for details. Retry with --force to force install it.
`
And here is the perl -v output:

`This is perl 5, version 24, subversion 1 (v5.24.1) built for MSWin32-x64-multi-thread

Copyright 1987-2017, Larry Wall

Perl may be copied only under the terms of either the Artistic License or the
GNU General Public License, which may be found in the Perl 5 source kit.

Complete documentation for Perl, including FAQ lists, should be found on
this system using "man perl" or "perldoc perl".  If you have access to the
Internet, point your browser at http://www.perl.org/, the Perl Home Page.

`

What do you think is causing this issue?

lordofhyphens, 2018-05-23:
No idea what is going on there behind the reported errors (perl is probably taking over via macros std::bind). However the build server doesn't fail.

Make sure your checkout of Slic3r is up to date

jclinton830, 2018-05-23:
I am using stable-1.1 branch. Do you think this is causing the issue? @lordofhyphens

lordofhyphens, 2018-05-24:
The "stable" branch isn't updated or supported, and the build instructions are only valid for 1.3.0 and newer on master branch.

lordofhyphens, 2018-05-24:
@jclinton830 it is a wiki, please edit it :)