slic3r/Slic3r #4396: Chamfer causes poor bottom solid layers

foreachthing, 2018-05-04:
### Version
Slic3r-master.2018.04.25.1667.bc145c5.64bit

### Operating system type + version
Windows 7 x64

### Behavior
[image]
The selected part (magenta) has chamfers. The other one is the same but without chamfers.

* When part has a chamfer at the bottom, the solid layers do some weird thing.
* Import attached stl and config.
* go to _Layers_ and select and compare layers 0.55, 0.75 and 0.95.

1. Part without chamfer - looks good (layer 0.95):
[image]

2. Part _with_ chamfer - not so good (layer 0.95):
[image]

#### STL/Config (.ZIP) where problem occurs
[UM2_BondtechMount.zip](https://github.com/slic3r/Slic3r/files/1973433/UM2_BondtechMount.zip)

xoan, 2018-05-04:
Probably related: https://github.com/slic3r/Slic3r/issues/3320#issuecomment-218104473

Tinchus2009, 2018-05-08:
Can also be elated to #4225 ? because te behaviour seen there actually hapens hen in top of the object there is a chanfer or a rounded border.

foreachthing, 2018-05-08:
@Tinchus2009 : I forgot about that! But, yes! That seems related, too. Funny thing is, I only have issues with the bottom layers...!?!?

dwillmore, 2019-01-13:
I can verify this behavior for both the top and bottom solid/perimeter problems.

dwillmore, 2019-01-15:
This is likely a dup of the earlier reports, but I'll wait to see what @lordofhyphens says.

lordofhyphens, 2019-01-15:
It probably is the same behavior as the other reported issues. It is a good test case though.