slic3r/Slic3r #4397: GSoC 2018 - Rotating Face to Bottom

lordofhyphens, 2018-05-04:
This issue is for @TheThirdOne to maintain his development log for his Slic3r GSoC 2018 project.

### User story:
---

**As** a user, I want to be able to rotate models so that a given face touches the build plate, **so** that I can manage plating difficult 3D models in Slic3r and not have to work out the math by hand.

---

This user story describes a wish described in #3047.

---

### Open Design Questions: 

If, after rotating and moving the model to put some face X onto the bed, another part of the model would be placed below the bed, what should be done?
* Error
* Shift model up in Z so that no part of the model is below the bed.

VanessaE, 2018-05-04:
Imho, if the attempted rotation puts part of the model below the plate, don't rotate, but do throw an error.

lordofhyphens, 2018-05-04:
My own thoughts on whether to push the model up or error is to push the model up in Z.
* The model is still oriented accordingly and support normally will be generated. 
* That the model isn't flush against the selected face is already an indicator to the user that what they want is impossible.
* Undo/Redo is available, so "mistakes" are recoverable in these situations.

dwillmore, 2018-05-04:
What is the purpose or this rotation/translation?
1) help orient the model so that it's more readily printable with the given technology
2) designate the 'bottom' of the model

For 1, then it makes sense to auto-add some kind of support/raft to the model or at least display a warning saying that the given facet is not a valid 'bottom'

For 2, orient it as comanded and let the part below the bottom hang.  Maybe throw an error, but orient the model as instructed.  It wouldn't hurt to display how much the model extends below the bed in case the user wants to make an educated decision as to what to do with the sub-bed portion.  It might be a rounding error that throws this error and the use would be wise to ignore it.  Or it might have escaped their notice and they need to deal with it some other way.

I don't like 1, but I think 2 doesn't do enough.  Maybe the error dialog could be ask "leave it alone and let Xmm hang below the bed", "auto-raise it up and add support/raft", "cancel".

foreachthing, 2018-05-04:
If I choose a certain face to be down, then for a reason. So, please, don't bother me with errors. Adjust the model in Z accordingly -> auto-rise, like @dwillmore  said.

MoonshineSG, 2018-05-05:
agree with @foreachthing. If the user select a face, it does so for a reason. A non-intrusive note/alert can be displayed to indicate that part of the model is below the bed. Highlight the "below the bed" in a different color to make it easily identifiable.

gege2b, 2018-05-05:
another possibility is to give user two menu entries (or whatever else the feature would be triggered)
- orient to bed (=> orient the face in direction of the bed and rise up the model if needed)
- orient and touch bed ( => orient the face to the bed and make this face the Z=0)

pro: everyone should be happy, no confirmation, no question, no warnings
con: two menu entries instead of one

qknight, 2018-05-15:
https://github.com/prusa3d/Slic3r/issues/147#issuecomment-389184624

TheThirdOne, 2018-05-15:
To summarize and to put my thoughts on the discussion about how underhang should be handled, it seems like most people are in favor of not erroring out, but at most giving a warning. There is not consensus about if the model should be moved up, kept where it is, or use two separate actions to do both.

My main concern with allowing part of the model to stay below the bed is that I don't believe any part of Slic3r can currently move things below the bed so that might be an assumption of various parts of the code. As a result, it might be a source of bugs; I may have missed a feature which allows you to position below the bed though. One option to combat that is to clip off the underhang, but that would  likely be a fiar amount more work to implement.

Yesterday I was able to get the start to a selection system working. It currently has two issues: it doesn't highlight the entire connected face just a triangle and its a bit slower than I would like. Here's a gif of it working.

[image]

TheThirdOne, 2018-05-15:
I like the idea of using a plane to note what face is going to be rotated. I think I would prefer highlighting the face all else being equal, but I hadn't considered drawing a new plane.

lordofhyphens, 2018-05-15:
Yeah, all you really care about is what plane gets used, so a slightly-offset plane parallel to the face would probably get the job done and would be easier to read, I think.

MoonshineSG, 2018-05-16:
@TheThirdOne if it helps to know, other slicers that have this function have single triangle instead of full face highlight. There's no "selection". When you click on the triangle it imediatly rotates the object.

And if the chosen face is not necesarily the one that is touching the bed, but it's the one facing the bed. Basically the user selects "down" and the models get's place on the bed with the "down"  parallel to the bed at the lowest posible Z, but not under the bed.  (hopefully this makes sense)

qknight, 2018-05-16:
@TheThirdOne as @MoonshineSG also sais, we probably don't need a **"permanent selection"** indicator, the **"mouse over selection"** is sufficient! i'd also recommend to stay with the **"triangle hoover"** as it makes programming easier and for most parts i assume it will just work as most often the problem is that a 'cut' surface, which wasn't aligned down needs to be rotated down and there the triangle surfaces will be quite big.

regarding the problem with the **selected triangle not being the lowest** one and making parts of the model overlap the zero plane you might have to come up with something you can actually implement. and later we'll see if that works in practice. 

maybe we should put together a list of thingiverse parts which act as an "example problem" so you can test the usability of your implementation.

* https://www.thingiverse.com/download:4634229
* https://www.thingiverse.com/download:4634241

MoonshineSG, 2018-05-16:
@qknight @TheThirdOne : I'm always of the opinion that having something that "just works" and later improve it is better than not having anything but "we are working on the best ever solution". 
Do someting, put it out there for people to try and give feedback and mostly remember... you can _never_ please everyone!

qknight, 2018-05-16:
@lordofhyphens 

> My own thoughts on whether to push the model up or error is to push the model up in Z.
> The model is still oriented accordingly and support normally will be generated.
> That the model isn't flush [...]

this is a brilliant idea as this technique can then also be used to align certain surfaces to the ground plate for cutting

lordofhyphens, 2018-05-16:
Minor thing to remember for later: There's color schemes in Slic3r and so the highlight color should be set up according to that system.

TheThirdOne, 2018-05-17:
I am aware. Pure red and blue were just easy to hardcode to help me work through it. I do plan on taking a second pass to fix things like the color scheme and decide how the UI should behave.

TheThirdOne, 2018-05-17:
I didn't make much visible progress today, but I am made progress getting the face normal and have begun work moving that value to the code to rotate (which is not complete).

mariusstrom, 2018-05-18:
FYI when you do this in Simplify3D (select a face to go on the plate that wouldn't be the lowest part of the model), S3D does what some described above: sets the face on the plate, then silently raises the model so that the new low point is actually on the plate. No warnings or error messages, and it's very intuitive to understand.

In addition, S3D also highlights the triangles/polygons rather than the full faces just as you've demonstrated above @TheThirdOne.

If it'd be helpful, I can do a quick video of how S3D handles this scenario.

TheThirdOne, 2018-05-18:
Today I got it to a point where the feature can actually be used. 

[image]

I tested it with the @qknight suggested, and there were some major framerate problems. When the model was first loaded it took several seconds to become responsive (presumably because I am populating a large array in perl). Additionally, even when not hovering over the model it was still slow. 

This shows two major optimizations that almost certainly will need to be made. Only looking up the face for the currently hovered model and perhaps loading the color buffer in CPP. 

Switching over to thinking about the UI. I found that on the d20, it was really easy to just use a triangle as an indicator, but that doesn't work on larger, more complex models. The triangles are simply too small. I didn't really have any indicator for what I was selecting.

Currently, the model will be pushed up so the lowest part of the model touches the bed. After looking through that code, I don't see an easy way to allow a part of the model to remain under the bed.

TheThirdOne, 2018-05-19:
I haven't fixed the startup freeze, but I have fixed the framerate after the first frame. I realized that I was copying the color data every frame, and fixing that made it a lot better. But with 10 complicated models, it was still quite slow. Making it so the face check only applies to the hovered model alleviated that problem. 

I then ran into what I think is a variation of #4420 where if I click the rotate face button three times it seg faults. I couldn't track down what the cause of the seg fault was though.

foreachthing, 2018-05-19:
Question(s): Would it not be more "natural" to select "Rotate Face" first, then the desired face? Because if I want to do something else, like scale the model, I'd end up with a selected face first; even though I want to scale the whole model.
... And you don't actually rotate the face, do you? ;-)

MoonshineSG, 2018-05-19:
Yes, select function then face

TheThirdOne, 2018-05-20:
I definitely agree that selecting "Rotate Face" then selecting the face is more natural, but I don't think it is technically feasible to to have it operate on the main 3d scene that way. It would require keeping the state of whether the "Rotate Face" button had been selected and disabling other features if it had. Though, if I can do it in a maintainable way, this is the obvious choice.

I am not sure whether the current method or the more natural way but in a new window (like cut and layer heights) is more natural though. One thing to note is that with the current solution, if there are other operations that need a face / plane, it would be easy for them to be added. If anyone feels strongly one way or the other, I would like to hear it. 

@foreachthing,  I'm not sure what you meant by "... And you don't actually rotate the face, do you? ;-)". If it was a joke about the label "Rotate face", a better label hasn't come to my mind, but I am very open to suggestions.

foreachthing, 2018-05-20:
@TheThirdOne, well no, not a joke! But you do rotate (home, align, drop, zero...) a BODY or model. If you edit a mesh, then you'd rotate a face...

robbycandra, 2018-05-20:
Hi... @TheThirdOne , your software above is interesting. Do you have any link to it ?

TheThirdOne, 2018-05-20:
@robbycandra, I've been working on [my fork of slic3r](https://github.com/TheThirdOne/Slic3r);

TheThirdOne, 2018-05-22:
Now face selection pops open a dialog.

[image]

I noticed that doing it this way also allows some additional options such as the plane to align the face with. Currently its not implemented, but its just a matter of adding in some more rotations behind an if. 

My next steps will be to work on the aesthetics of face selection. At a minimum, I will be aligning the hover and selection colors with Slic3r's color scheme, but I will probably try to use a plane like what's in the cut dialog.

lordofhyphens, 2018-05-22:
@TheThirdOne in your dialog, could you suppress the color highlight for the entire object and use that for the face selection? Since it should be the only model there, it should be expected to re-use the select color for just a face. Saves you from having to find another contrasting color.

lordofhyphens, 2018-05-22:
@TheThirdOne It's moving along enough that I think you can (and should) open a pull request against Slic3r/Slic3r soon. That will start exercising some of the build servers and let me take a closer look at your work and start the review process when you indicate it's ready to be looked at seriously.

TheThirdOne, 2018-05-22:
@lordofhyphens: About the selection color, that was the plan. I'll open a pull request when I go through the dialog a bit more (so probably end of tomorrow).

TheThirdOne, 2018-05-23:
I implemented the selection color stuff that was discussed. I took a quick look to see what it would take to reuse the plane rendering used in the cut dialog; that code not reusable for planes not already aligned with the axes. 

I then started working on making it so that ctrl-z doesn't have to be hit twice to undo it. Currently it does because I am just doing two rotations under the hood.

There are two ways that I could fit it into the current undo/redo system:
  1. Make it so rotations can operate about arbitrary axes (rather than just X,Y, or Z)
  2. Make an operation that groups two elementary operations so that be undone and redone together.

For this feature, 2 allows more flexibility, but I think that having 1 might allow some new features to be easier to implement. For example, rotating an object about the normal from the face might be useful and would be easy following from the code I have already done + option 1. I don't know exactly what 2 might be used for, but it seems generally useful. 

I started checking out how 2 would need to be implemented, because from what I can tell 1 requires a lot of modification to do.

foreachthing, 2018-05-23:
Suggestion: Now you can rotate a face down onto the plate... how about an option where you can rotate a face up? Maybe I need _this_ face to be extra-clean but can't be on the plate. Or maybe this face is off by just a few degrees but needs to be parallel with Z. Just a thought!

MoonshineSG, 2018-05-23:
Isn’t there a “flip” function in Slic3r ?

foreachthing, 2018-05-23:
@MoonshineSG, well, you're right! First, align the face to the plate, then rotate it (around  X or Y) by 180°.

TheThirdOne, 2018-05-24:
I implemented the group undo internal feature. It works swimmingly. I was able to get the dialog plane select to actually work as well. 

However, I kept hitting crashes do to thumbnail stuff. So I spent some time working on #4420 because without some fixes, the extra rotations guarantee a crash with rotating to align with the xz or yz planes.

TheThirdOne, 2018-05-25:
Spent more time investigating 4420. I made a quick fix to make it so that this feature doesn't crash, but I haven't solved the issue really. 

All that is really left before this is complete it finishing up the UI. So what should this feature be called on the toolbar? Drop plane and rotate face are two options that have been previously mentioned. Additionally, what should the icon for this look like?

lordofhyphens, 2018-05-25:
"Rotate to face" for a name, perhaps?

mariusstrom, 2018-05-25:
Simplify3D calls this "Place surface on bed", menu-exposed only (no icons).

gege2b, 2018-05-25:
just a quickly made icon based on @lordofhyphens idea  (full size and 16x16)
oh and this is not a octahedron :D 

[image]
[image]

VanessaE, 2018-05-25:
Way too much detail for the size, imho.  How about a tetrahedron, rotated off-axis a bit (so that no face looks like is points straight down) instead?

VanessaE, 2018-05-25:
Something like this maybe?
256px:
[image]

32px: [image]

xoan, 2018-05-25:
New icons should be based on [Silk Iconset](http://www.famfamfam.com/lab/icons/silk/) to maintain the current look and feel.

Probably by using a mix between some block icon with a colored face and an arrow.

PS. Not at home at this moment, so I can't do it now.

TheThirdOne, 2018-05-29:
I did a bit of work on top of @VanessaE's idea. As it was it took up too much space within the 32x32 to fit in with the other icons. So I scaled it down, moved the arrow out to the right and changed the colored face to red. Then I added a blue plane to the bottom. 

256px:
[image]

32px: [image]

In the toolbar (colored):
[image]

In the toolbar (uncolored):
[image]

I think this is pretty close to done. I think the colors could be adjusted a bit to fit in better, and size (and arrangement) could be modified a bit too. I also think it might be better without the black lines; they don't come through well when scaled down. However, I would to get some feedback from yall before I continue.

lordofhyphens, 2018-05-31:
This will be in the next round of builds on https://dl.slic3r.org/dev

gege2b, 2018-06-01:
I gave it a quick try and it seems to work as expected

the only problem I can see is the icon witch is twice the size (compared to the others) on windows 10

[image]

gege2b, 2018-06-01:
I think some tweaks can be done on the face selection screen : 

- Make the "rotate to plane" button enabled only if a face is selected. With some models, faces are really tiny and are difficult to see. Making the button behave this way can be a visual feedback when a face is effectively selected

- Trying to make selected face more visible for the same reason (but actually I don't really know how - maybe with some edge emphasis on the selected face ? )

qknight, 2018-06-01:
regarding 'make selected face more visible': maybe just draw a transparent plane which would be the new bottom plane after rotation? not sure if that works well but at least a idea worth trying.

gege2b, 2018-06-01:
Arbitrary plane has been discussed in the thread, ( [here](https://github.com/slic3r/Slic3r/issues/4397#issuecomment-389276987) and [here ](https://github.com/slic3r/Slic3r/issues/4397#issuecomment-391197498) )

TheThirdOne, 2018-06-01:
@gege2b I accidentally made the icon 32x32 rather than 16x16 which doesn't look bad on Linux. Making the rotate button only enabled when a face is selected is a good idea too. It should be an easy to add.

TheThirdOne, 2018-06-02:
I implemented both of those suggestions that I mentioned above.

[image]

TheThirdOne, 2018-06-05:
So I took a look at #4435, and it turns out that it is my fault. I forgot to set the color when rendering quads.

The fix is simple; I've already done it.

lordofhyphens, 2018-06-05:
#4438 needs addressing.

7eggert, 2018-09-12:
I just tried the feature in slic3r-1.3.1-dev-d17659b-linux-x64. It works as I would expect it, but it would be better to use the main window since the popup is too small.

The current interface is OK. Maybe (or not) offering thee buttons instead of one dropdown + button would improve usability. Pro: Easier to align to XZ /YZ. Con: Easier not to align to XY by accident.

I agree that by selecting a non-top face, e.g. the inner part of a digit of the d20, the model should be moved up and a message should be displayed - I'd prefer before cutting.

Other users have mentioned that they'd like a message or would use the feature for cutting the object, too. Then it would be nice to display the amount of lifting being done, either before actually performing the cut or in a message box: "The selected face was automatically raised to z=4.711 mm in order to allow printing" (write that down, open cut menu, ...)

If people would _commonly_ like to cut the model on faces, you could include the cut button in that dialog, too (that's just an idea, not a request).

TL;DR: ☺, also the icon is nice.

TheThirdOne, 2018-09-12:
>  It works as I would expect it, but it would be better to use the main window since the popup is too small.

What exactly are you envisioning? In the early stages of development it did use the main window, but running face selection constantly (for a rotate to plane button) has problems and having a face selection mode requires a fair amount of plumbing. I ended up just matching the cut dialog for consistency. 

One thing I think would be an improvement would be switching the popup into where the main 3d view is (for both cutting and rotate face). Not sure how feasible that is, but that seems like close to the ideal UI.  

> Maybe (or not) offering thee buttons instead of one dropdown + button would improve usability. Pro: Easier to align to XZ /YZ. Con: Easier not to align to XY by accident.

Not sure which of the two is better. Definitely an option though.

> Then it would be nice to display the amount of lifting being done, either before actually performing the cut or in a message box: "The selected face was automatically raised to z=4.711 mm in order to allow printing" (write that down, open cut menu, ...)

For technical reasons, giving a message about the amount lifted would be slightly hard.

7eggert, 2018-09-13:
> One thing I think would be an improvement would be switching the popup into where the main 3d view is (for both cutting and rotate face). Not sure how feasible that is, but that seems like close to the ideal UI.

This↑.

We are at the 3D plater, I press the button. The right part [print settings] … changes to the "rotate face" interface, the buttons at the top are grayed out (except rotate face, which is pressed down and will cancel  the operation on unpressing, too) and the main area is replaced with the 3d face selection screen and no other objects but the selected one.

It's nothing where I'd say: "I can't be bothered to maximize the window, do that for me", but it would be nice.

ast0815, 2020-03-14:
What ever happened to this feature? Did it get implemented? If so, where do I find it in the UI?

VanessaE, 2020-03-15:
It was implemented.  You can find it in the toolbar or the object's right-click menu.

closed 2020-03-15 by VanessaE