slic3r/Slic3r #4398: Slic3r Perl prebuilt dependency archive location.

EitanSomething, 2018-05-04:
### Version
Current github Slic3r repository
commit version 8a93fbb
****

windows 10
version 1709

I don't know where to put the custom Perl package of 5.24 prebuilt dependency archive

lordofhyphens, 2018-05-05:
/path/to/slic3r/local-lib

You can always also download a prebuilt archive of Slic3r from dl.slic3r.org and extract it over the source checkout.

closed 2018-05-05 by lordofhyphens

EitanSomething, 2018-05-06:
So I should create a file local-lib in the copy of slic3r and then run 
perl Build.PL
perl Build.PL --gui

slate0, 2019-01-13:
My location is: "D:\myname\Documents\dev\slicer\Slic3r\local-lib" where local-lib is the folder contained in the download.  When I run perl Build.PL I get a message saying that the script couldn't find them after about 15 minutes of work.  I'd like to understand what I don't understand about your instructions.

lordofhyphens, 2019-01-13:
@slate0 follow the wiki instructions for getting and building Boost with mingw. Then set the BOOST_DIR environment variable to the directory.