slic3r/Slic3r #4399: Very long Diagonal Line

N3X15, 2018-05-05:
# Version
```
Rob@RAGE C:\Users\Rob\Downloads\slic3r
> Slic3r-console.exe --version
1.3.0-dev
```

Extracted from Slic3r-master.2018.04.26.1669.8a93fbb.64bit.zip, which probably means git commit 8a93fbb.

# Environment
* Windows 7 Ultimate x64 SP1

# Behavior

In Slic3r build 2018.04.26.1669 (8a93fbb?), when the application is slicing a Benchie using my printer settings, it generates an extremely long stepped diagonal line that exits the bounds of the print plate (which is interesting on its own).  

## Reproduction Steps
1. Import config.ini, as provided in the ZIP.
2. Load the provided Benchie STL in Plater.
3. Check the preview.

## Expected Behaviour
A nice little boat with a 6mm skirt.

## Observed Behaviour
Expected boat, and a line stretching to the horizon.

[image]
[image]

## Notes
I have had this issue on-or-off previously, but this is the first time I've observed it consistently persisting after a restart of Slic3r.

I just rebuilt the extruder on the printer and experiment with wax paper as a print surface, so some of my settings may be a little weird.

# STL/Config (.ZIP) where problem occurs
* **NOTE:** GitHub is being a butt and won't let me upload the ZIP, so it's externally hosted on my dedicated server.

https://www.nexisonline.net/downloads/bug-reports/slic3r/slic3r-diagonal-line.zip

N3X15, 2018-05-05:
And this time, I only noticed after the printer had made a mess: 
[image]

lordofhyphens, 2018-05-05:
> For the record, I fixed this by setting brim connections width to 0.
> Still, it should:
>
>    - Not print outside of the build plate bounding box.
>    - Not try to connect to things that don't exist.

FYI the bed size is just a note to the user, we don't enforce it.